pavel-demin/red-pitaya-notes #1011: VNA Client Server Protocol double check

Sguerrault, 2021-06-30:
Hello Pavel,

We started using a red pitaya recently in one of our research project, and more especially your VNA application.
It works great and will actually allow us to perform measurements in the field, which would have been impossible previously. Thanks for that !

I spend some time reading the different server and client codes you provided over the last few years for the VNA app (C/Python/Matlab), in order to better understand the protocol, as we would like to build a Labview client.

I think I got the protocol right : 

### Example to send the sweep size value to the server :

- Use the value 2 and left shit it by 28 bits (or multiply 2 by 2^28).
- Add the desired size (let's say 300)
- The result is 536871212.
- Cast the result in uint32.
- Open a connection to the server IP on the port 1001.
- Write the result above on the port ((4 bytes written).

### Example to trigger the sweep :

- Use the value 10 and left shit it by 28 bits (or multiply 10 by 2^28).
- The result is 2684354560.
- Cast the result in uint32.
- Open a connection to the server IP on the port 1001.
- Write the result above on the port (4 bytes written).

Is that correct or did I really missed the point ?
When I implement this in Labview, the commands are sent, but no triggers occur (I connect / send a start value, a stop value, a sweep size and then trigger the sweep). I wonder if that's because I did not send all the parameters ? (levels etc). Is that mandatory if I first connect with the python client. I'd assume the server keeps the value from the previously connected client, whatever It may be ?

Thank you in advance for your insights.
Sébastien

pavel-demin, 2021-06-30:
Hello Sébastien,

It's great to know that the VNA application is useful for you.

Your description of the protocol is correct.

It is not necessary to send all the parameters. The default levels are correct and the server keeps the parameters from the  previously connected client.

However, if your client disconnects after starting the sweep, the sweep stops immediately. The connection should be kept open until the sweep is complete. This connection is also used to read the data.

Best regards,

Pavel

pavel-demin, 2021-06-30:
I see that it's possible to use .NET libraries from LabVIEW. Maybe an easier solution would be to write a .NET library that communicates with the VNA server. Here is a link to a similar .NET library that I wrote for the pulsed NMR application:

https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/pulsed_nmr/client/PulsedNMR.cs

Sguerrault, 2021-06-30:
Hello Pavel,

Thanks for the diligence of your answers and confirming our general understanding of the protocol is correct.

### About the .net idea : 
Indeed, one can call .NET libraries from Labview (used to be one of the easiest way to implement some complex Matlab processing tool in Labview, not simple scripts). 

However, I find it surprising that one have to resolve to this workaround. I myself built hybrid TCP client server systems in the past (Labview/Labview or Labview/.Net) and there never really was any issue. 

I'll double check if our code is not at fault here before trying a potential .Net workaround.

Would you mind keeping the issue open, as I think I'll have some questions when It comes to reading the data after the sweep ?

Best regards,
Sébastien

pavel-demin, 2021-06-30:
> Would you mind keeping the issue open, as I think I'll have some questions when It comes to reading the data after the sweep ?

OK. Let's keep it open.

Sguerrault, 2021-06-30:
I actually found the origin of the issue. That was a big endian / little endian thing that seems to be a bit different in Labview. After conversion, it works (both passing parameters and triggering the sweep).

I am now reading the values, but struggling a bit with the conversion from the received bytes string to a usable array. Any advice on this matter ? Regarding the size, I should read 1200 bytes for a 300 sweep size, shouldn't I ?

Best regards,
Sébastien.

pavel-demin, 2021-06-30:
> Regarding the size, I should read 1200 bytes for a 300 sweep size, shouldn't I ?

There are four floating point values per sweep point: re1, im1, re2, im2. So, the total size of the data is 300 * 4 * 4 = 4800 bytes.

Sguerrault, 2021-07-01:
Hello Pavel, 

Just to make sure I understand the structure correctly :

For each sweep point, a quadruplet is generated, i get that part. So that means the first 1*4*4 : 16 bytes define the quadruplet for the first sweep frequency ? So the first 4 bytes are Re1, then comes the 4 for Im1, 4 for Re2 and 4 for Im2 ?

So the string I receive as to be restructured, as you did in your matlab example with the reshape function. You then obtain a 4 columns by 300 lines array, each line representing a frequency point.

Or did i get it totally wrong and the first 1200 bytes represent the 300 values of Re1, the next 1200 ones the 300 values of Im1 ... etc ?

Thanks in advance!
Sébastien.

pavel-demin, 2021-07-01:
> So the first 4 bytes are Re1, then comes the 4 for Im1, 4 for Re2 and 4 for Im2 ?

Yes, this is correct.

Re1 and Im1 are from the first RF input (IN1).

Re2 and Im2 are from the second RF input (IN2). They are currently not used in the Python code.

> So the string I receive as to be restructured, as you did in your matlab example with the reshape function. You then obtain a 4 columns by 300 lines array, each line representing a frequency point.

Yes, this is correct.

It is of course possible to read 4 floating point values (16 bytes) at a time and repeat this enough times to read all frequency points.

Sguerrault, 2021-07-03:
Hello Pavel,

It finally works ! Once again, an issue of bytes order in LabVIEW. 

I have 2 final questions : 

- I calculated the phase as 180/pi*arctan(Im1/Re1). However, it feels like I do not get the same results with the python client and mine. Looks like there's a factor 2 somewhere ?

- I noticed you generate a table of frequencies ( self.freq = np.linspace(start, stop, size)) and use them later on. However, those frequencies likely do not match exactly the ones used during the sweep ? Or are you generating them exactly the same way on the board so you expect those 2 arrays of frequency values to match ? 

Best regards,
Sébastien.

pavel-demin, 2021-07-03:
My answer to the two questions: I do not know why you think it does not work, it works well for me.

If you want more detailed answers, please provide more detailed questions with examples illustrating the problems.

pavel-demin, 2021-07-04:
> I calculated the phase as 180/pi*arctan(Im1/Re1)

Looks like the correct function to use is `atan2(Im1, Re1)`:

https://zone.ni.com/reference/en-XX/help/371361R-01/glang/inverse_tangent_2_input/

Sguerrault, 2021-07-04:
Hello Pavel, 

Thank you very much for pointing that out. There's indeed two different atan functions in Labview. I tried the other one and I then obtain the same result as your python client.

About the other question, my apologies if that was unclear. My reference point is your python code so I obviously blame any discrepancies on my own LabVIEW code. 

I was basically trying to make sure I understand how you reconcile the actual frequency table (as generated by the FPGA and swept on the RF output) and the one on the customer side. 

- Client side : np.linspace(start, stop, size
- FPGA side : the same thing ?

I actually encountered something else. When I start trying to adjust the decimation/bandwith, it seems it get very slow and the read function easily times out. Do you see any reason why this could happen ? There is no problem if I adjust the bandwith via the python software then trigger etc from my own code. It seems to be the only parameters that causes such a thing.

Thanks for your assistance.

Once completed, I'd like to share the labview code for anyone else to use as their own basis.

Sguerrault, 2021-07-04:
I think I am getting confused between the rate and the bandwidth :

self.rateValue.addItems(['5000', '1000', '500', '100', '50', '10', '5', '1'])

vs 

rate = [10, 50, 100, 500, 1000, 5000, 10000, 50000][value]

Am I missing some conversion between decimation and bandwidth somewhere ?

pavel-demin, 2021-07-04:
> Am I missing some conversion between decimation and bandwidth somewhere ?

bandwidth in Hz = 125e6 / 2500 / rate

where rate is the configurable decimation rate

Sguerrault, 2021-07-04:
> > Am I missing some conversion between decimation and bandwidth somewhere ?
> 
> bandwidth in Hz = 125e6 / 2500 / rate
> 
> where rate is the configurable decimation rate

Thanks Pavel, I add actually just figured It out a moment ago.
I think all works just fine now, the topic can be closed :).

Once again thank you for your assistance !

closed 2021-07-04 by Sguerrault