pavel-demin/red-pitaya-notes #1013: vna.py points limit and minor bug

rus-ace, 2021-09-10:
### Description of the setup:

 * Device:Red Pitaya
 * SD card image:bit stream generated from vna_128_88
 * Application:vna.py
 * Other relevant information:

### Description of the problem:
1. Not really a problem but what limits the number of points in vna.py to 32768. Can this be easily increased?
2. Issue with vna.ini
The start and stop frequencies do not update the graph axes if saved in vna.ini even though they show correctly. That is, if I change stop frequency to 500kHz and write config file, when i re-open vna.py 500 kHz shows in the Stop,kHz window but the graph plots to 5000kHz.

### Steps to reproduce the problem:

 1.open vna.py, set stop frequency to 500kHz, run a scan, write cfg file. close vna.py
 2.open vna.py, read cfg file, run a scan
 3.

pavel-demin, 2021-09-10:
Thank you for your interest in the VNA project.

> Not really a problem but what limits the number of points in vna.py to 32768. Can this be easily increased?

Currently, all the frequencies are stored in a memory buffer in the FPGA before starting the scan. The size of this memory buffer is 32768.

It is possible to increase the number of points by storing them in the DDR3 memory and updating the memory buffer in the FPGA during the scan.

What size are you looking for?

> Issue with vna.ini

Thank you for finding this problem. I will try to reproduce and fix it in the coming days.

pavel-demin, 2021-09-11:
> That is, if I change stop frequency to 500kHz and write config file, when i re-open vna.py 500 kHz shows in the Stop,kHz window but the graph plots to 5000kHz.

Could you please clarify what graph does not changes as you expect?

> 1.open vna.py, set stop frequency to 500kHz, run a scan, write cfg file. close vna.py
> 2.open vna.py, read cfg file, run a scan

I have tried to reproduce the problem but without much success.

If you change the stop frequency and only run a single DUT scan, then only the stop frequency of the DUT data changes. The stop frequency of the calibration data does not change unless you redo calibration.

So, if you change the stop frequency and only run a single DUT scan, the x axis of the graphs from DUT data to Gain open changes. The Open, Short and Load data graphs do not change unless you redo calibration.