pavel-demin/red-pitaya-notes #1015: Feature request - FST4W mode for WSPR transceiver

dc1rdb, 2021-09-14:
Hi Pavel, 
AI6VN is about to incorporate a simultaneous decode all of the FST4W modes into his wsprdaemon. Is there any chance something similar could be done for your WSPR transceiver?
Thx,
Robert

pavel-demin, 2021-09-14:
Hi Robert,

I see that wsprdaemon uses the JT9 decoder for the FST4W mode and I cannot find a detailed technical description of this mode.

If the sample rate of the FST4W mode is not 375 samples per second, then it would require a different FPGA configuration.

So, I do not see how the FST4W mode can be added to the WSPR application for Red Pitaya. I think a separate application should be created to work with this mode.

Best regards,

Pavel

dc1rdb, 2021-09-14:
Thanks for your quick response, Pavel!
Let me see if I can find some detailed description for you.

73,
Robert

pavel-demin, 2021-09-14:
Some ideas about the sample rates can be found in the code at [this link](https://sourceforge.net/p/wsjt/wsjtx/ci/master/tree/lib/fst4_decode.f90). Looks like the sample rate for FST4W-120 is 12000/205 samples per second.

I do not think the FST4W-120 mode could be added to the WSPR application for Red Pitaya.

closed 2021-09-14 by pavel-demin