pavel-demin/red-pitaya-notes #1019: WSPR Transceiver: verify correct operation of GPS interface

dc1rdb, 2021-10-05:
Hi Pavel,
this is just a question rather than an issue:
Is there any way via console to verify that a GPS module (connected via DIO3_N and UART) is operating correctly for the time synchronization and for the automatic measurement and correction of the frequency deviation?

Thanks in advance!
73 de Robert

pavel-demin, 2021-10-06:
Hi Robert,

With the connected GPS UART interface, the following command enables the GPS time synchronization:
```
service gpsd start
```

The time synchronization can be checked using one of the chrony commands. If the GPS module is used for the time synchronization, the following command should show some information about the GPS time source:
```
chronyc sources
```

The following commands could be used to check the frequency correction:
```
cat apps/sdr_transceiver_wspr/sdr_transceiver_wspr.bit > /dev/xdevcfg
apps/common_tools/measure-corr 30
```
The returned value is the frequency deviation in ppm.

Best regards,

Pavel

dc1rdb, 2021-10-06:
Thanks for the quick response, Pavel.
Here are my results:
```
rp-f08971:~# service gpsd start
 * Bringing up network interface lo ...
ip: RTNETLINK answers: File exists                                        [ ok ]
 * Starting gpsd ...                                                      [ ok ]

rp-f08971:~# chronyc sources
MS Name/IP address         Stratum Poll Reach LastRx Last sample               
===============================================================================
#? PPS                           0   4     0     -     +0ns[   +0ns] +/-    0ns
^+ time.cloudflare.com           3   9   377   251   -257us[ -263us] +/-   12ms
^- pmg.jonasled.de               2   8   377   129   +350us[ +338us] +/-   33ms
^* mail.klausen.dk               2   9   377    66   +216us[ +200us] +/-   12ms
^- gromit.nocabal.de             2   9   377   263   +272us[ +267us] +/-   29ms
^+ wpad.fritz.box                3   6   377    17   +178us[ +178us] +/-   26ms

rp-f08971:~# apps/common_tools/measure-corr 122.88 30
3.99

```
It does not seem that GPS is being used for time sync, does it?
Frequency correction seems to be working, however.
Is it possible to include the `service gpsd start` into the start.sh script?

Thx and best regards
Robert

pavel-demin, 2021-11-19:
The fixed version of the frequency correction scripts is now in the [new release](https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20211119).

pavel-demin, 2022-03-22:
> It does not seem that GPS is being used for time sync, does it?

The fixed version of the gpsd configuration is now in the [new release](https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20220322).