pavel-demin/red-pitaya-notes #1020: Build Xilinx kernel and apply real-time patch PREEMPT_RT

ChristianDoll95, 2021-10-16:
Hello Pavel,

first I want to thank you for your very helpful and impressing support for the Red Pitaya Community. 

I have taken over the project from my former colleague, which was described in the issue #981.
https://github.com/pavel-demin/red-pitaya-notes/issues/981 

Because I wanted to reduce the latency of the whole system, I have switched the project to your Linux Alpine v3.14 platform. 
In the meantime this works also with cross debugging. 

I measure the time for reading out a signal frame from the SD Card and write it to a RAM buffer, where the DMA (PL side) takes it from. 

System: Red Pitaya STEMLAB 125-14

Measurement time: 90 minutes
Latency with Debian Jessie (and also after Upgrade to Debian Stretch): typ. 80..200 µs, max. > 1 ms
Latency with Alpine v3.14: typ. 40..80 µs, max. 280 µs 

So the Alpine system already reduces the latency strongly. But for my application (as part of my master thesis project) it would be very nice to get under 200 µs or even 100 µs maximum latency. 

I think this should be possible with the PREEMPT_RT real-time patch. 

So I have tested to build a kernel myself under Ubuntu (with cross compiler for the Zynq), because my understanding is, that on Alpine a kernel patch is only possible before building the kernel. 
After a few modifications, this worked for me with the following commands.

Used tutorial: [https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842481/Build+kernel](https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842481/Build+kernel) 

Used commands:
make ARCH=arm64 xilinx_zynqmp_defconfig
make ARCH=arm64 menuconfig         # here I have set a few modifications
make all CC=arm-linux-gnueabihf-gcc ARCH=arm UIMAGE_LOADADDR=0x2080000 uImage

Now the problem is as follows: I have only found the sources for the Kernel 5.10.0-xilinx. 
[https://github.com/Xilinx/linux-xlnx](https://github.com/Xilinx/linux-xlnx)
After writing the uImage file to the SD card, the Red Pitaya boots and works well, but I do not get any module connected. 
So the whole Board Support Package (BSP) and especially the modules in "modloop" are not fitting to my kernel version. 

If I list the connected modules with **lsmod**, the list is empty. 
In the debug messages, which I can see with **dmesg | less**, I get the following messages for a few packages (pps_gpio, spi_dev, 8021q and 80221).

e.g.: pps_gpio: disagrees about version of symbol module_layout

Some of the .ko-Files, which are missing, are part of the build process of the uImage File for 5.10.0-xilinx kernel. But not all, and also not all can be found by the kernel with lsmod after importing them to the modloop file (packed as .sqsh-file and written again to the sd card). 

My three questions are: 
1. Do you know how to fix this issue, that the uImage of version 5.10.0-xilinx does not accept the modules (stored in modloop) of version 5.10.46-xilinx?
2. Otherwise do you know a link to download the sources of a newer xilinx-kernel versions?
3. Do you think, if it is even worth to do the PREEMPT_RT patch on this kernel to improve performance and reduce latency? If yes, a tutorial would be very nice.

Many thanks in advice,
Christian

pavel-demin, 2021-10-16:
Hello Christian,

I never tried the PREEMPT_RT patch but I think it can be added to my Makefile.

My Makefile downloads the Linux source code from www.kernel.org.

I have just updated the scripts and Makefile and set the Linux version to 5.10.73. So it should be possible to apply the patch and rebuild the SD card image with the following commands:
```
git clone https://github.com/pavel-demin/red-pitaya-notes
cd red-pitaya-notes
make tmp/linux-5.10
curl -LO https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/5.10/patch-5.10.73-rt54.patch.gz
zcat patch-5.10.73-rt54.patch.gz | patch -d tmp/linux-5.10 -p 1
source helpers/build-all.sh
```

Best regards,

Pavel

pavel-demin, 2021-10-16:
I have just found some instructions at [this link](https://wiki.trenz-electronic.de/display/PD/How+to+install+the+linux-rt+%28Real-Time%29+patch) and tried to set some kernel parameters (fully preemptive kernel, 1000 Hz timer frequency, disable the CPU frequency scaling) as recommended in these instructions.

pavel-demin, 2021-10-16:
The rtl8188eu driver does not work with the PREEMPT_RT patch:
```
  CC [M]  drivers/net/wireless/realtek/rtl8188eu/core/rtw_cmd.o
In file included from ./include/linux/spinlock.h:328,
                 from ./include/linux/mmzone.h:8,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/umh.h:4,
                 from ./include/linux/kmod.h:9,
                 from ./include/linux/module.h:16,
                 from drivers/net/wireless/realtek/rtl8188eu/include/basic_types.h:13,
                 from drivers/net/wireless/realtek/rtl8188eu/include/drv_types.h:15,
                 from drivers/net/wireless/realtek/rtl8188eu/core/rtw_cmd.c:6:
drivers/net/wireless/realtek/rtl8188eu/include/osdep_service_linux.h: In function '_exit_critical_ex':
drivers/net/wireless/realtek/rtl8188eu/include/osdep_service_linux.h:244:25: error: passing argument 1 of 'rt_spin_unlock' from incompatible pointer type [-Werror=incompatible-pointer-types]
  244 |  spin_unlock_irqrestore(plock, *pirqL);
      |                         ^~~~~
      |                         |
      |                         _lock * {aka struct raw_spinlock *}
./include/linux/spinlock_rt.h:104:44: note: in definition of macro 'spin_unlock'
  104 | #define spin_unlock(lock)   rt_spin_unlock(lock)
      |                                            ^~~~
drivers/net/wireless/realtek/rtl8188eu/include/osdep_service_linux.h:244:2: note: in expansion of macro 'spin_unlock_irqrestore'
  244 |  spin_unlock_irqrestore(plock, *pirqL);
      |  ^~~~~~~~~~~~~~~~~~~~~~
./include/linux/spinlock_rt.h:25:51: note: expected 'spinlock_t *' {aka 'struct spinlock *'} but argument is of type '_lock *' {aka 'struct raw_spinlock *'}
   25 | extern void __lockfunc rt_spin_unlock(spinlock_t *lock);
      |                                       ~~~~~~~~~~~~^~~~
```
Looks adding this patch requires more work than I initially thought.

ChristianDoll95, 2021-10-17:
Hello Pavel, 

thank you very much for your fast reply, the time consuming tests and the very helpful description! 

I will test it again the following days. Because I do not use the wireless connection, maybe I can live without that drivers. 

I will give you feedback as soon as possible. 

Best regards,
Christian

closed 2021-11-02 by pavel-demin