pavel-demin/red-pitaya-notes #1022: Phase coherency sdr_transceiver_122_88

glenenglishgithub, 2021-11-11:
Hi Pavel
Great work !

I am  trying to use the sdr_transceiver_122_88 as a diversity receiver , and the RX LO needs to be phase coherent for both RX chains.

I have the development machine setup and I can generate the bit files OK (I write VHDL with Vivado in project mode  normally)
I put a single dds in the top-level (block_diagram) and used an axi stream  broadcaster to split the stream and pipe it to both rx in trx blocks....but my understanding of your scripts and tcl is limited I cannot get the syntax right.

Also, because the DDS output is valid each clock cycle, I feel we do not need any handshaking and so the AXI stream broadcaster is not necessary
Maybe you can find time please to provide a RX phase coherent (single frequency, single LO for both RX0 and RX1 chains ) version of sdr_transceiver_122_88

best regards,
glen english

pavel-demin, 2021-11-11:
Hi Glen,

The problem is that sdr_transceiver_122_88 was never intended to have phase coherency. There are some other projects in this repository that are easier to make RX phase coherent.

For example, the sdr_transceiver_hpsdr_122_88 project has already a possibility to enable the RX phase coherency and is confirmed to work as a diversity receiver. Basically, the diversity receiver functionality works out of box with PowerSDR mRX PS. If the TX part is not needed, then the sdr_receiver_wide_122_88 project would be much easier to modify by replacing multiple DDS blocks with one.

Best regards,

Pavel

glenenglishgithub, 2021-11-11:
Hi Pavel
thank you for the reply. Thank you for the idea to look at the hpsdr project. 
Today I have  learned your TCL scripts, and got to learn (more ) XIlinx TCL,  and I have it working .  I piped the same dds_data_tdata to both cpx muls with no timing issues. 122 MHz is easy for 7 series of course with narrow busses.
Once again, your efforts to generate many SDR projects , great documentation , and very good build detail and instructions are very much appreciated. 
best regards, glen english

closed 2021-11-11 by glenenglishgithub