pavel-demin/red-pitaya-notes #1024: update-corr.sh not calling measure-corr with right parameters for 125 or 122.88 ?

waterwin, 2021-11-18:
Using Red Pitaya 125-14 with latest image from this source.
I was trying to start the frequency correction by connecting a GPS. This has worked for me before (as in a year ago or so).

update-corr.log shows:
Thu Nov 18 15:19:01 UTC 2021
Usage: measure-corr freq time
 freq - frequency expressed in MHz (122.88 or 125),
 time - measurement time expressed in seconds (from 1 to 30).
It took me some time to understand this but I did see their was no correction done via the CORR parameter.

When checking update-corr.sh you can see only the TIME parameter is set as 30'
The FREQ parameter is not set so the system is not knowing whether you use the 125-14 or 122.88-16

This might be something left over after integrating the 122.88 and 125 images.

I will try to correct my update-corr.sh and post after that again.

pavel-demin, 2021-11-18:
Thank you for finding this problem. It was recently discussed in #1019 and I think it is now fixed with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/b8ec0e16b808d0c944a2bea6c1cfb2b04c974670).

pavel-demin, 2021-11-19:
The fixed version of the scripts is now in the [new release](https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20211119).

closed 2021-11-19 by pavel-demin