pavel-demin/red-pitaya-notes #1026: RP keeping tracking of current frequency to another radio via CAT commands

ik3umt, 2021-12-07:
Hi Pavel, 

I don't know if you're a radio amateur , anyway, would it be possible to keep a commercial radio (ICOM in this case) tracked to RP frequency via CAT protocol commands ?
I mean, while changing frequency with HPSDR software , RP generates same frequency changes CAT commands to be sent via serial interface to another radio in order to keep both radios always on the same frequency ?

Thank you

Rolf-L, 2021-12-07:
Hi Federico,
I'm using HPSDR + RP and CAT Commands to control other software like JTDX etc. 
The CAT Control commands are limited and support only TS-2000, TS-480, TS-50S and PowerSDR commands.
I dont know, if your ICOM support one of this commands.
73s de Rolf -DJ7TH- 
www.dj7th.de

ik3umt, 2021-12-08:
Rolf, 
I'm new to RP and HPSDR, I just seen PowerSDR has CAT commands compatible with Kenwood radios, perhaps an easier way is to make a KenwoodCAT to IcomCat converter rather than implement it into RP and work just at PC serial line side.
thank you
Federico
ik3umt

Rolf-L, 2021-12-08:
Hi Federico,
the CAT commands from Kenwood, Icom and Yaesu are totaly different.
A CAT converter programm running in the background of Windows could be great help.
It should read and write the serial ports and convert the commands which are stored in a convert table.
Would be a nice job for a software writer :-)
73s de Rolf -DJ7TH-

ka6s, 2021-12-08:
A couple of points.  The radio software is a client in that it responds to
commands. It doesn't typically originate - but reacts to received commands
on the CAT channel.  The Kenwood TS-2000 is a popular command set because
it contains so much functionality.  I wrote the PiHPSDR CAT software - so
have spent a lot of time in this space.  What you need is some sort of T -
software that copies the commands from the application and can translate
specific commands (FA command for the HPSDR software) and send the results
two places I think, i.e. back to the original application and to the second
radio.

Steve KA6S

ja5aea, 2022-01-14:
Hi,

I may misunderstand the question raised by ik3umt, but I understand "void icom_write()" routine of current "sdr-transceiver-hpsdr" program on RP keeping of current band to another radio via cat commands, if he connects RP and Icom radio with RP serial interface. 

If he changes CAT command from band to frequency in the software, he accomplishes his desire.

Am I wrong?

take

de JA5AEA

closed 2022-03-22 by pavel-demin