pavel-demin/red-pitaya-notes #1027: Feature request - include some version info in WSPR spots upload

dc1rdb, 2021-12-08:
Hello Pavel, the wspr.rocks database is supporting version / source information of the uploaded spot data, see attached screenshot, last column.
Would it be possible to add something similar to the spot data uploaded from RedPitaya running your multiband WSPR transceiver software?
Thanks in advance for your consideration.
73 de Robert

[image]

pavel-demin, 2021-12-08:
Hello Robert,

I was recently informed that version information can be sent to wsprnet.org by adding `-F version=VERSION` to the curl command in [decode-wspr.sh](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wspr/app/decode-wspr.sh#L65).

Best regards,

Pavel

dc1rdb, 2021-12-08:
Great, thank you. Do you have any preference on how your software should identify itself?

pavel-demin, 2021-12-08:
No, I do not have any preferences. Since this version field is optional and it is not used by wsprnet.org, I think you can fill it in with any information you want. I do not think it should be used to identify my software.

dc1rdb, 2021-12-08:
Works perfect, thx!

closed 2021-12-08 by dc1rdb