pavel-demin/red-pitaya-notes #1030: Alpine Image working with RP Z7020 SOIC

mgraeser, 2022-01-16:
### Description of the setup:

 * Device: RP Stem-Lab 125-14-Z7020
 * SD card image: Own image based on your Alpine linux image
 * Application: Own developmen
 * Other relevant information:

### Description of the problem:
I wanted to test the new RP version based on the bigger FPGA (Z7020). I have a working image for the Z7010 where I omitted any FPGA image load routines while starting up. The Z7010 version boots normally and I can access it via SSH.
I expected that the 7020 version would work fine as long as I do not try to load up a bitfile designed for the other chip. However, the system does not boot at all  using the same SD Card. The card remains unresponsive.

I just wanted to ask if my assumption that the linux image should work on both is correct. 

Thank you for your time an consideration. 
Matthias

pavel-demin, 2022-01-16:
> I expected that the 7020 version would work fine as long as I do not try to load up a bitfile designed for the other chip.

Yes, your assumption is correct. It is exactly how my SD card image works on both STEMlab 125-14 (Z7010) and STEMlab 122.88-16 SDR (Z7020). I had to remove bitstream from boot.bin to make it work with these two cards.

mgraeser, 2022-01-18:
Thank you. I did not think of the boot.bin. That explains it. Your image works fine, so I will see to it that we will adapt the image from scratch using your scripts. Thank you for the fast help!

closed 2022-01-18 by mgraeser