pavel-demin/red-pitaya-notes #1033: Hi Rolf,

dl6jan, 2022-02-04:
Hi Rolf,

> Is there a RP VNA version available, using IN2 and OUT2 instead of IN1 and OUT1?.

Using IN2 requires a very simple modification of the Python code in vna.py: `adc1` should be replaced with `adc2` [on line 212 in vna.py](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/vna/client/vna.py#L212).

Using OUT2 requires some modifications of the FPGA configuration.

_Originally posted by @pavel-demin in https://github.com/pavel-demin/red-pitaya-notes/issues/170#issuecomment-317254189_

Hi Pavel,
I'd also like to use your VNA software with IN2 and OUT1 or OUT2. Long ago you wrote, the vna.py program can be modified at line 212 by replacing adc1 with adc2. The current version of vna.py looks completely different so I can not find something like that on line 212.
Is still possible to use IN2 instead ??  On line 593 - 605 I find the following:

 adc1 = self.data[0::2]
 adc2 = self.data[1::2]
 attr = getattr(self, self.mode)
 start = self.sweep_start
 stop = self.sweep_stop
 size = self.sweep_size
 attr.freq = np.linspace(start, stop, size)
 attr.data = adc1[0:size]copy()
 self.update_tab()
 self.reading = False
 if not self.auto:
 self.progressBar.setValue(0)
 self.set_enabled(True)

Maybe  "attr.data = adc1[0:size]copy()" is the line that had to be changed ???

Can you help ?

Regards Christian (DL6JAN)

pavel-demin, 2022-02-04:
> Maybe "attr.data = adc1[0:size]copy()" is the line that had to be changed ???

Yes, I think it should be enough to replace `adc1` with `adc2`:
```
        attr.data = adc2[0:size].copy()
```

To enable OUT2, it should be enough to set 'Level 2' to 0 dB in the VNA GUI.

dl6jan, 2022-02-04:
Hi Pavel,
thanks a lot for your fast reply, will try it out in the next days ...
What I am still missing is a manual for your vna.py program. I'm not an expert and I do not even know what "Level2" in your VNA GUI is but think I'm able to change it to 0dB. Most of the GUI is self explaining but I do not understand all of the functions/buttons I see there .... 
Anyway, I'm willing do study it :-)   

Chris, DL6JAN

closed 2022-03-22 by pavel-demin