pavel-demin/red-pitaya-notes #1034: FT8 TX options

Pricey42, 2022-02-08:
Hi Pavel,
Thanks for creating this software for the Red Pitaya SDR.  I have been using the FT8 transceiver to receive signals on multiple bands.  Is there a way to use the device to transmit?  Ideally I'd like to send a free form text FT8 message.  If this is possible how do I do it?

Thanks Mark

pavel-demin, 2022-02-09:
Hi Mark,

The FT8 transmission is currently incomplete and the software part is missing. The main problem is that I do not know what to transmit. The FT8 protocol does not provide beacon mode and implementing interactive mode is too much work for me.

If you have an idea of a fixed message that could be transmitted periodically without disturbing other FT8 operators, then I think it could be implemented similarly to [transmit-wspr-message.c](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wspr/app/transmit-wspr-message.c).

Another possibility to transmit FT8 messages using the Red Pitaya board is to combine the following elements:
 - [SDR transceiver compatible with HPSDR](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-hpsdr/)
 - [PowerSDR mRX PS](https://github.com/TAPR/OpenHPSDR-PowerSDR)
 - [virtual audio cable](https://github.com/rdp/virtual-audio-capture-grabber-device)
 - [WSJT-X](https://www.physics.princeton.edu//pulsar/K1JT/wsjtx.html)

Best regards,

Pavel

Rolf-L, 2022-02-09:
Hi Mark,

As Pavel suggested, I've been using this combo for years. 
This has the advantage that changes in a part can be updated at any time without any problems.

73s de Rolf -DJ7TH-
[www.dj7th.de](url)