pavel-demin/red-pitaya-notes #1035: Bitstream built from source always get redpitata crashing

olc, 2022-02-13:
### Description of the setup:

 * Device: **SDRlab 122-16**
 * SD card image: **red-pitaya-alpine-3.14-armv7-20211119**
 * Application: **sdr_transceiver_122_88**
 * Other relevant information: **Xilinx_Unified_2020.2_1118_1232**

### Description of the problem:

The red pitaya always crashes when using the `sdr_transceiver_122_88` bitstream built by myself.

### Steps to reproduce the problem:

 1. Install a development machine following the [Development machine](http://pavel-demin.github.io/red-pitaya-notes/development-machine/) notes.
 2. Build the `sdr_transceiver_122_88` bitstream from source following the instructions at [SDR transceiver](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-122-88/): 
```
source /opt/Xilinx/Vitis/2020.2/settings64.sh
git clone https://github.com/pavel-demin/red-pitaya-notes
cd red-pitaya-notes
make NAME=sdr_transceiver_122_88 PART=xc7z020clg400-1 bit
```
 3. Make filesystem writable on the red-pitaya: `rw`
 4. Copy the generated bitstream to the red pitaya:
```
scp tmp/sdr_transceiver_122_88.bit root@192.168.1.33:/media/mmcblk0p1/apps/sdr_transceiver_122_88/
```
 5. Reprogram the FPGA with the new generated bitstream : `cat sdr_transceiver_122_88.bit > /dev/xdevcfg`
 6. Start the server on IN1 (after killing any previous one if present): `./sdr-transceiver 1` : the red-pitaya does immediatly crash (electrical reboot required).

Note that size of my generated bitstream is by far different from the size of the original one:
```
-rwxr-xr-x    1 root     root       1164168 Feb 13 16:34 sdr_transceiver_122_88.bit              # generated by myself
-rwxr-xr-x    1 root     root       2619572 Feb 13 13:34 sdr_transceiver_122_88.bit.origin   # original bitstream
```
Do you see what could be wrong?

Thanks in anticipation for any help!

pavel-demin, 2022-02-13:
The file size is indeed wrong and I cannot reproduce the problem using the commands from the notes. I have just tried to rebuild `sdr_transceiver_122_88.bit` and its size is 2619572.

I think the smaller file size indicates that the part specified during the build was not `xc7z020clg400-1`.

I have just tried to build `sdr_transceiver_122_88.bit` with the following **wrong** command:
```
make NAME=sdr_transceiver_122_88 bit
```

The size of `sdr_transceiver_122_88.bit` in this case is 1164168. So, looks like you are using the wrong command to build 
`sdr_transceiver_122_88.bit`.

So please check what command you are using to build `sdr_transceiver_122_88.bit` and try to use the following correct command: 
```
make NAME=sdr_transceiver_122_88 PART=xc7z020clg400-1 bit
```

olc, 2022-02-13:
I have `rm -rf tmp/*` and then run the correct command again as you suggested and the magic happened: file size is now  2619572!

I should have done it earlier but I thought that deleting the .bit could be enough (the full rebuild takes more than 30 minutes on my setup).

Many thanks for your help, Pavel.

closed 2022-02-13 by olc