pavel-demin/red-pitaya-notes #1037: Question about FT8 Skimmer

pjsg, 2022-02-24:
@pavel-demin I'm thinking about building a multiband PSK31 skimmer based on the ft8 skimmer setup. My feeling is that with a small number of these scattered around the world, then we could capture *all* the PSK31 traffic on the standard bands. I run pskreporter.info nd I suspect that the total bandwidth increase required to do this PSK31 skimming would be small. In particular, I'm thinking about emergency comms -- initially unidirectional from the crisis area. By getting all the traffic into a single location it can then be distributed more widely to other interested parties. 

Does this seem crazy? How much effort was required to make the "application" part of the ft8 decoder?

Thanks

pavel-demin, 2022-02-25:
I did some quick reading on the PSK31 protocol. Looks like the bands are organized quite similarly to FT8. However, the transmissions are not started synchronously at a well-defined instant. So the decoding should be continuous more similar to CW skimming. Please correct me if I miss something.

In the FT8 application, most of my effort has gone into adapting the open source FT8 decoder from the WSJT-X project and developing an uploader for pskreporter. Other parts were taken from the WPSR application with minor changes.

I think the key component for such a project is a decoder, preferably open source.

pjsg, 2022-02-25:
I had found: http://www.moetronix.com/ae4jy/pskcoredll.htm where there is
downloadable LGPL licensed C++ library that does PSK decoding.

Then there is also the psk code from fldigi:
http://www.w1hkj.com/files/fldigi/fldigi-4.1.20.tar.gz   -- look in the
src/psk directory. This is GPL3 and again written in C++

The second code looks more modern, and has a nice function: `*int* psk::
rx_process(*const* *double* *buf, *int* len)` which processes the audio
samples for a specific audio passband. It looks as though this are the real
part of the samples, though the code converts into complex as it multiplies
by each of the sub carriers.

Unfortunately I'm not a DSP expert (or even a DSP anything). I know enough
to talk to experts!

Thoughts?

Philip

pjsg, 2022-02-27:
I wanted to use the Red Pitaya board itself. They make a very convenient (and reliable) monitoring system. I have found that using SparkSDR with (say) an HL-2 does not stay running for more than a few days without something happening (typically to the windows box). If the FPGA firmware is the same as for the FT8 skimmer, then that would be ideal as it then it might be possible to run both in the linux engine. 

I'd like to make it simple for existing people who run FT8 skimmers based on your code on the RP boards to be able to choose which bands they want to skim PSK31 from. 

I'm only using 10 channels on my StemLab122-16. 

There are 18 callsigns running 'STEMlab SDR FT8 TRX 1.0' and a further 23 running 'Red Pitaya FT8 TRX 1.0'

VK5ARG, ES5PC, AF7KR, YU/G8ZMF, BA7KW-2, DG9BJA, NA5WH, D4Z, JI1HFJ, YO8SEP, DL5PF-RP14, OE9GHV, VE7CC, DR4W, DM6AS, N6TV, 9M2CNC, J68HZ, DD6UWE, ZR1TH, EA8BFK, ZL4YL, DJ9PC, N1DQ, F1EYG, N0KWJ, DL9GTB, F4VTQ/1, N7TUG, K6FOD, OH0K/6, DE0QPG, F4VTQ/2, SM7IUN, ZF1A, KC2IEB/2, DD9LH, SM6PPS, HB3YIQ, OH6BG, 9V1HP

While I haven't plotted these on a map, there is a reasonable amount of worldwide coverage.