pavel-demin/red-pitaya-notes #1038: Alpine will not boot on a new 122-16 SDR Pitaya

mgarrett1955, 2022-02-28:
* Device:
 * Pitaya 122-16 SDR
 * SD card image:
 * red-pitaya-alpine-3.14-armv7-20211119 2
 * Application:
 * sdr_tranciever 
 * Other relevant information:
 *

 I have started with a fresh card, deleted all partitions, created a new blank partition and copied the files onto the card.
The board light up the green LED, but does flash any other LEDS, and does not boot, There is no Ethernet activity out of the board.

The board works fine with the supplied Red Pitaya application, (Which does not allow installing your applications like the 125-14 does).

I have performed this same operation many times on the 125-14 boards.

Thank you in advance, the work that you have done here is amazing..!

Michael

pavel-demin, 2022-02-28:
Green LED on is OK. No other LEDs on is also OK.

Please check the boot process using USB/serial console:
https://redpitaya.readthedocs.io/en/latest/developerGuide/software/console/console/console.html

As far as I remember, most of the time this kind of problem is solved by replacing the SD card.

pavel-demin, 2022-02-28:
Yes, there is a web interface that can be used to start the applications.

The Ethernet interface is by default configured to request an IP address via DHCP. If no IP address is provided by a DHCP server, then the Ethernet interface falls back to a static IP address [192.168.1.100](http://192.168.1.100/).

Some additional information on network configuration can be found at [this link](http://pavel-demin.github.io/red-pitaya-notes/alpine/#network-configuration).

mgarrett1955, 2022-03-01:
Ok, chrome had blocked the site, fixed that, now the web page is being served.. all up and running. Thank you again!!

Michael

pavel-demin, 2022-03-02:
I am glad that it finally works for you.

closed 2022-03-02 by pavel-demin