pavel-demin/red-pitaya-notes #1039: Cant connect GRC on Windows (Radioconda)

JensZander, 2022-03-04:
### Description of the setup:

 * Device: 122-16
 * SD card image: red-pitaya-alpine-3.14-armv7-20211119
 * Application: sdr_transceiver_122_88, GRC
 * Other relevant information: Windows, Radioconda GRC build

### Description of the problem:

The Red Pitaya Souce block is not able to connect to RP application

_Traceback (most recent call last):
.....
..... red_pitaya.py", line 44, in __init__
  fd = os.dup(self.data_sock.fileno())
OSError: [Errno 9] Bad file descriptor_

Connecting to Alpine (pTTy) works fine.  Running PowerSDR with sdr_receiver_hpsdr_122_88  also works fine so the hardware seems OK.
Is there some Linux/WIndows incompatibility ?

### Steps to reproduce the problem:

 1. Compile and run trx_am.grc  on GRC
 2.
 3.

pavel-demin, 2022-03-04:
> Is there some Linux/WIndows incompatibility ?

Yes, the way red_pitaya.py handles sockets and file descriptors is incompatible with Windows.

If you want to use the 122-16 board with GNU Radio on Windows, then I think that the easiest solution is to use the applications compatible with HPSDR and the following GNU Radio block:

https://github.com/Tom-McDermott/gr-hpsdr

Another solution is to use Windows Subsystem for Linux (WSL) to run GNU Radio:

https://wiki.gnuradio.org/index.php/WindowsInstall#WSL_|_Ubuntu

pavel-demin, 2022-03-04:
I have just realized that [gr-hpsdr](https://github.com/Tom-McDermott/gr-hpsdr) also requires Linux.

So, looks like using WSL is the only solution.

JensZander, 2022-03-05:
Thanks Pavel - will try the WSL-solution.
/jens

JensZander, 2022-03-11:
I went to native linux (Ubuntu 20.4) and now I got the connection working with GR 3.10.1.1.  
A minor issue - there seems to be some incompatabilty in the "template" flowgraphs trx_am and trx_ssb:
The resampler box could not handle float expressions för the Interpolation/Decimation parameters (fixed with int(..) )
In the trx_ssb flowgraph The FFT filter tap description 'firdes' produces an error 
   ' type object 'gnuradio.filter.filter_python.firdes' has no attribute 'WIN_HAMMING'
Looking at the Gnuradio dicussion threads this seems to be version related - will be looking to find a fix to this.

JensZander, 2022-03-11:
Fixed with the new "Band-pass filter taps" block in GRC 3.9 och 3.10