pavel-demin/red-pitaya-notes #1042: WSPR app and gpio in

davidhay45, 2022-04-22:
Question

Is it possible to read a gpio_in pin while the wspr app is running.  I suppose I am asking if the sdr_transceiver_wspr.bit supports read access to the gpio spare pins?

pavel-demin, 2022-04-22:
Currently there are no general purpose inputs implemented in the sdr_transceiver_wspr project. It is possible to use an I2C I/O expander (PCA9555, MCP23017, etc.) or a microcontroller for additional inputs and outputs.

davidhay45, 2022-04-22:
Hi Pavel

Hope you are well and glad to see the project is thriving.

The reason for my question: I am trying to interface the wspr app to an ATU100 (David N7DDC) to set the necessary tune parameters for TX using I2C.  I plan on using a gpio output to signal the ATU to look for an I2C command but I need to know when the ATU is ready.  So an I2C I/O expander is a good idea if that is the only way.

Thanks

closed 2022-04-27 by davidhay45