pavel-demin/red-pitaya-notes #1043: Vivado 2019.1 problem

davidhay45, 2022-04-27:
### Description of the setup:

Windows 10 with the Ubuntu app and Vivado 2019.1

### Description of the problem:

I am trying to create the necessary development machine to be able to "play" with fpga.

I tried to update my 2017 Vitualbox to run Debian 9.13 but encountered problems that I concluded were due to windows 10 issues.  So instead I loaded the Windows 10 Ubuntu app.  I had not realised that Vitis 2020.2 was the suggested software and have instead loaded 2019.1.
The led_blinker make ended with the following:
<<<
INFO: [Common 17-206] Exiting Vivado at Wed Apr 27 11:14:02 2022...
mkdir -p tmp
/opt/Xilinx/Vivado/2019.1/bin/vivado -nolog -nojournal -mode batch -source scripts/hwdef.tcl -tclargs led_blinker

****** Vivado v2019.1 (64-bit)
  **** SW Build 2552052 on Fri May 24 14:47:09 MDT 2019
  **** IP Build 2548770 on Fri May 24 18:01:18 MDT 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

source scripts/hwdef.tcl
# set project_name [lindex $argv 0]
# open_project tmp/$project_name.xpr
Scanning sources...
Finished scanning sources
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1700] Loaded user IP repository '/opt/Xilinx/Xilinx_Vivado_SDK_2019.1_0524_1430/red-pitaya-notes/tmp/cores'.
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/opt/Xilinx/Vivado/2019.1/data/ip'.
# write_hw_platform -fixed -force -file tmp/$project_name.xsa
invalid command name "write_hw_platform"
    while executing
"write_hw_platform -fixed -force -file tmp/$project_name.xsa"
    (file "scripts/hwdef.tcl" line 6)
INFO: [Common 17-206] Exiting Vivado at Wed Apr 27 11:14:15 2022...
make: *** [Makefile:145: tmp/led_blinker.xsa] Error 1
redpitaya@Vostro470:/opt/Xilinx/Xilinx_Vivado_SDK_2019.1_0524_1430/red-pitaya-notes$
>>>

I should explain that initially the system would not find the vivado executable so I changed the Makefile to include the full vivado path.
The led_blinker.bit is created and is a valid bit file in that it works on my 125-14 but my question is: what is the write_hw_platform "error"?

Should I update to Vitis 2020.2 (or later)?

Thanks

pavel-demin, 2022-04-27:
> I should explain that initially the system would not find the vivado executable so I changed the Makefile to include the full vivado path.

Normally, the PATH environment variable is configured by the following Vivado setup command:
```
source /opt/Xilinx/Vivado/2019.1/settings64.sh
```
This command must be executed before make.

> The led_blinker.bit is created and is a valid bit file in that it works on my 125-14 but my question is: what is the write_hw_platform "error"?
> Should I update to Vitis 2020.2 (or later)?

Some Vivado commands have been replaced by other commands in Vivado 2019.2. My scripts are currently compatible with Vivado 2019.2 and newer versions.

The commands related to the generation of bitstream files have not changed. So, the script that generates the bitream files still works with older versions of Vivado.

If you only want to create the bitstream files, you can do that with Vivado 2019.1. The corresponding make command is
```
make NAME=led_blinker bit
```
If you want to build the SD card image, then you will need to install Vivado 2020.2.

Some of the projects in this repository use the DSP48 IP core that is not available in newer versions of Vivado. 2020.2 is therefore the most recent version that can be used to build all projects in this repository.

pavel-demin, 2022-05-06:
Have you tried to download and install [Xilinx Vitis 2020.2: All OS installer Single-File Download](https://www.xilinx.com/member/forms/download/xef.html?filename=Xilinx_Unified_2020.2_1118_1232.tar.gz).

sudo apt-get --no-install-recommends install \
  build-essential bison flex git curl ca-certificates sudo \
  xvfb fontconfig libxrender1 libxtst6 libxi6 lib32ncurses5 \
  bc u-boot-tools device-tree-compiler libncurses5-dev \
  libssl-dev qemu-user-static binfmt-support zip \
  squashfs-tools dosfstools parted debootstrap zerofree
apt-get install \
  build-essential bison flex git curl ca-certificates sudo \
  xvfb fontconfig libxrender1 libxtst6 libxi6 lib32ncurses5 \
  bc u-boot-tools device-tree-compiler libncurses5-dev \
  libssl-dev qemu-user-static binfmt-support zip \
  squashfs-tools dosfstools parted debootstrap zerofree

davidhay45, 2022-05-06:
That is the file Xilinx_Unified_2020.2_1118_1232.tar.gz which is the one that crashes on install.

pavel-demin, 2022-05-17:
Using Debian 9 is OK.

I think the problem in make.txt can be solved by running 
```
source source /opt/Xilinx/Vitis/2020.2/settings64.sh
```
before running make.

The `arm-linux-gnueabihf-gcc` compiler is provided by Vitis 2020.2.

closed 2022-05-17 by davidhay45