pavel-demin/red-pitaya-notes #1046: /dev/cma and /dev/mem drivers installation possibility on RedPitaya 125-14 board with Linux Ubuntu OS

popovkon, 2022-05-15:
### Description of the setup: RedPitaya 125-14 board with default Linux Ubuntu OS

 * Device: RedPitaya 125-14
 * SD card image: Linux Ubuntu
 * Application: adc_recorder_trigger
 * Other relevant information: 

### Description of the problem:
Dear Pavel,
Thank you very much for your work and very useful projects with RedPitaya boards!
I work with a project where I need to digitize data from both ADC channels of RedPitaya 125-14 board. The digitization starts with external trigger. After that, I need to send this data to the Host computer via EPICS protocol. EPICS is Experimental Physics and Industrial Control System software. Version of EPICS is R3.14.x.  EPICS R3.14.x can not be installed on the Alpine Linux because of architecture incompatibility. 
1) Is it possible to install /dev/cma and /dev/mem drivers on Linux Ubuntu?
I need to use AXIS_RAM_WRITER_V1_0 ip core in my project.

2) If it is possible, please could you explain the installation steps of the drivers on Linux Ubuntu?

Thank you so much in advance for your time and help!

Sincerely,
Popov Konstantin

pavel-demin, 2022-05-15:
I think it should be possible to build the cma module using the following Makefile:
```
obj-m += cma.o
KDIR = /lib/modules/$(shell uname -r)/build
all:
        make -C $(KDIR)  M=$(shell pwd) modules
clean:
        make -C $(KDIR)  M=$(shell pwd) clean
```
The current version of the cma module can be compiled for Linux 5.10. For other versions of Linux, it is necessary to modify some includes.

Alternatively, it should be possible to replace the cma module with the [udmabuf](https://github.com/ikwzm/udmabuf) module that offers more functionality.

I think /dev/mem should already be present on your system and nothing needs to be done to enable it.

popovkon, 2022-05-16:
Dear Pavel,
Thank you for your response!

Yes, I already have the /dev/mem module on my system.

I checked the Kernel version.

I have default Linux Ubuntu from RedPitaya.
Kernel version is 4.9.0.
Which includes do I have to modify to compile the cma module on the 4.9.0 Kernel version?
I am a beginner with drivers work. Excuse me, if I ask noob questions.

Sincerely,
Popov Konstantin

pavel-demin, 2022-05-16:
> Which includes do I have to modify to compile the cma module on the 4.9.0 Kernel version?

I do not know. I would try to compile the module and check for error messages.

Do you need Ubuntu from RedPitaya because of some other applications or just to be able to install .deb packages?

If you just need the packages, you can try using my SD card image based on Debian 9:

https://www.dropbox.com/sh/5fy49wae6xwxa8a/AAB1xH748EwOday_5ZN24nsva/red-pitaya-debian-9.13-armhf-20210423.zip?dl=1

The system in this SD card image has the cma module installed.

I also have an SD card image based on Debian 11 that I plan to release soon.

popovkon, 2022-05-17:
Dear Pavel,
Thank you so much for your response and explanation!

I need the Linux OS to be able to install .deb packages.
I have tested Debian based image.
EPICS software can be easily installed on the Debian Linux image.

Thank you very much for your help!
My problem is solved now.

Spasibo!

Sincerely,
Popov Konstantin

closed 2022-05-17 by popovkon

TanviWani, 2025-04-29:
Respected Sir, 
I am a BTech student trying to get data from red pitaya. I want to modify the adc_test but my Linux is missing kernel. I have loaded the latest image of the SD Card and operating on Linux 5.15. Can you help me the CMA drivers. I have gone through another post of Pavels where he mentioned creating a CMA driver, but i am inexperienced and cannot follow how am i supposed to use it. The debian image that is shared cannot be accessed and requires permission. Hoping to get some help. Thank You.

pavel-demin, 2025-04-29:
If you have a new question, please create a new issue or email me. However, first, please read my notes at [this link](http://pavel-demin.github.io/red-pitaya-notes/). The information that you are looking for may already be there.

I do not think that discussing new questions in comments on old, closed issues is a good idea. Doing so risks spamming the original issue authors.