pavel-demin/red-pitaya-notes #1047: SDR Tranceiver Packet Structure

bitsandgigas, 2022-05-18:
Hi Pavel,

Thanks for providing some awesome projects for the red-pitaya. I am working on my thesis and designing communication scheme for underwater acoustics. I've setup your sdr_tranceiver project and have it running. I have matlab set up and I'm able to receive the tcp packets, but I can't figure out the structure. 

- Does each packet contain the i and q ? if so how is it structured?
- Am I looking at this wrong? Is the data sent over tcp already demodulated? I was under the impression it was just decimated and the i and q were seperated so the sdr software would do the demod.

When i have it figured out I would be willing to share my matlab code for connecting for connecting to the sdr.

pavel-demin, 2022-05-18:
IQ samples are sent as pairs of floating point numbers in single precision format. I think the following Matlab code should be able to read them:
```
% connect to server
t = tcpclient('192.168.1.100', 1001);

% set center frequency in Hz
freq = 3000000;
write(t, uint32(0 * 2^28 + freq));

% set sample rate (20000:0, 50000:1, 100000:2, 250000:3, 500000:4, 1250000:5)
rate = 2;
write(t, uint32(1 * 2^28 + rate));

global size;
size = 10000;

function sdr_callback(t)
global size;

% read data
data = read(t, 2 * size, 'single'); 

% convert data
data = reshape(data, 2, size);
re = data(1, 1:size);
im = data(2, 1:size);
iq = complex(re, im);

% process data
% ...
end

configureCallback(t, 'byte', size, @sdr_callback)
```

pavel-demin, 2022-05-19:
I do not see why the frequencies cannot be alternated with the SDR transceiver application if the timing requirements are not very strict.

If the timing requirements are strict, then you will need an FPGA configuration similar to [VNA](http://pavel-demin.github.io/red-pitaya-notes/vna/).

pavel-demin, 2022-05-19:
> So i can't detect different frequencies at the same time
> everything's under 100 khz

If all of your signals are below 100 kHz, you probably do not need to change the center frequency for each packet. I would suggest trying to set the sample rate to 100 kSPS and the center frequency to 50 kHz and do all the modulation and demodulation in Matlab.

willsull14, 2022-05-20:
> I do not see why the frequencies cannot be alternated with the SDR transceiver application if the timing requirements are not very strict.
> 
> If the timing requirements are strict, then you will need an FPGA configuration similar to [VNA](http://pavel-demin.github.io/red-pitaya-notes/vna/) where the frequencies are alternated by the FPGA.

Timings not so much an issue more thinking how to know what frequency to receive on at what time. The point is to be noncoherent and not need the transceivers synchronized to real time.

bitsandgigas, 2022-05-20:
> I do not see why the frequencies cannot be alternated with the SDR transceiver application if the timing requirements are not very strict.
>
> If the timing requirements are strict, then you will need [...]

The timing isn’t so much an issue it’s more that I need to know what frequency to receive on at a given time because the devices need to be noncoherent and not need to be synchronized to the same time. So once I detect a preamble I know the order of frequencies but I need to be monitoring on all channels until then.

bitsandgigas, 2022-05-20:
> If all of your signals are below 100 kHz, you probably do not need to change the center frequency for each packet. I would suggest trying to set the sample rate to 100 kSPS and the center frequency [...]

I wasn’t sure if the FIR filtered out low frequencies. So I’ll set it to 50 kHz I’ll use 250 ksps so I can get more resolution. And I’ll be able to receive up to 100 khz.

I’m a little confused on what happened to the I and a if the frequencies are set to 50 khz. If I’m receiving on say 15khz would I combine i and q back together and then demodulate? Now that I’m saying it I think that’s what your telling me lol.

pavel-demin, 2022-05-20:
> So I’ll set it to 50 kHz I’ll use 250 ksps so I can get more resolution.

The idea is to set the center frequency to half the sample rate to cover the spectrum from 0 to the frequency equal to the sample rate. So, for 250 kSPS, the center frequency would be 125 kHz.

I suggest connecting SDR# to the SDR transceiver application and playing around with different sample rates and center frequencies. This way you will see for yourself what settings work for your project.

As I have answered all your questions, I am closing this issue.

closed 2022-05-20 by pavel-demin

bitsandgigas, 2022-05-20:
Thank you! Appreciate the help!