pavel-demin/red-pitaya-notes #1048: Question about platforms

davidhay45, 2022-05-19:
Hi Pavel 

I see that the recommended development platform is now Debian 11.3, are you intending to make a pre built SD card image also using Debian 11.3?

Also, what about the Alpine pre-built?

Thanks

pavel-demin, 2022-05-19:
I recently updated the scripts that build the Debian based SD card images with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/a56e8f4399207720432c854f7ddd2638b595fe76). The SD card image linked from the notes is still based on Debian 9. I do not update this SD card image very often because it is quite a long process. You can build it following the instructions at [this link](http://pavel-demin.github.io/red-pitaya-notes/led-blinker/#sd-card-image).

> Also, what about the Alpine pre-built?

I am not sure what the question is.

davidhay45, 2022-05-19:
Is it better to use Debian 9/11 on the red-pitaya instead of the Alpine?

Thanks