pavel-demin/red-pitaya-notes #1049: Development machine - critical errors question

davidhay45, 2022-05-23:
### Description of the setup:

 * Other relevant information: My development machine is running on Debian 11.3 under VBox on Windows 10 and using Vitis 2020.2

### Description of the problem:

Are "critical errors" important?  For example when I make the following:
make NAME=led_blinker_122_88 PART=xc7z020clg400-1 bit
I get "56 Warnings, 39 Critical Warnings and 0 Errors encountered." in the impl_1 log and "85 Warnings, 0 Critical Warnings and 0 Errors encountered." in the synth_1 log.
Do the critical errors matter?  The bit file works ok.

BTW When installing Debian and Vitis I had to use a 192 GB vdi and then compact it after install using:
cd /
dd if=/dev/zero of=zfile bs=1M
rm –f zfile

And then vboxmanage modifyhd <full path>debian.vdi –compact from the vbox program directory on Win 10.

I also found I had to define a portforward on Adaptor 1 of 3022-->22 to enable SSH to work.

pavel-demin, 2022-05-23:
I have just checked the 39 critical warnings of the led_blinker_122_88 project and they are all about pin assignments to signals that do not exist in the project. A solution would be to replace [cfg/ports.xdc](https://github.com/pavel-demin/red-pitaya-notes/blob/master/cfg/ports.xdc) that is used by all the projects with a separate .xdc file for each of the projects but I prefer to the current solution.

> Do the critical errors matter?

There is no critical errors in the led_blinker_122_88 project. The critical warnings seems to be OK.

davidhay45, 2022-05-23:
OK, critical warnings not errors.

Many thanks.

closed 2022-05-23 by davidhay45