pavel-demin/red-pitaya-notes #1051: Question: How do the discrete sampling rates come about?

dxxx9, 2022-06-15:
Hi Pavel,

I've started using a RedPitaya STEMlab 125-10 and came across your implementation in this repo. First of all thanks for your work!

I tested the standard `sdr_transceiver` you've written and wanted to ask why you only allow those discrete sample rates below:

```python
rates = {20000: 0, 50000: 1, 100000: 2, 250000: 3, 500000: 4, 1250000: 5}  # L. 31 from sdr_transceiver/gnuradio/red_pitaya.py
```

I explicitly wondered why the maximum sample rate is given as 1250kHz. Because due to [the specs](https://redpitaya.com/stemlab-125-10/) the max rate should be 10 times higher.

Thanks :)
Daniel

pavel-demin, 2022-06-15:
BTW. In the [SDR Transceiver](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver) notes there is a diagram showing how the sample rates are changed by the various blocks in the signal processing chain.

pavel-demin, 2022-06-15:
I do not understand your questions.

dxxx9, 2022-06-15:
The maximum data rate of the CPU is lower than I expected. The reason you gave in your first answer is the limited FPGA resources. I just wanted to ask again more exactly how this maximum frequency of the CPU of 1250 kHz comes about (in the [diagram](https://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver/) you referred to). 

I stumbled on this because for my use case this CPU rate is just the limiting factor.
Thanks a lot for your helpful feedback!

[image]

pavel-demin, 2022-06-15:
> I just wanted to ask again more exactly how this maximum frequency of the CPU of 1250 kHz comes about (in the [diagram](https://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver/) you referred to).

I do not know what to add to my first answer. The process is very simple:
   - find good round sample rates that can be obtained from 125 MSPS (6.25, 3.125, 2.5, 1.25 MSPS)
   - select the maximum sample rate that can be achieved with the available FPGA resources

> I stumbled on this because for my use case this CPU rate is just the limiting factor.

Maximizing sample rate was never the goal of the [sdr_transceiver](https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/sdr_transceiver) project. There are probably ways to optimize its resource usage and increase the sample rate.

There are some other projects in this repository that have higher sample rates ([sdr_receiver_wide_122_88](https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/sdr_receiver_wide_122_88), [SDR transceiver for STEMlab 122.88-16 SDR](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-122-88/), [wideband SDR transceiver](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-wide/)).

dxxx9, 2022-06-16:
Ok, thanks again for the clarification! 👍

closed 2022-06-16 by dxxx9