pavel-demin/red-pitaya-notes #1052: Question: How to modify project for the STEMLab 125-10?

EggertEnjoyer123, 2022-06-30:
I am trying to build a improved acquisition system for the Red Pitaya. I'm particularly interested in having more samples for the spectrum analyzer functionality, and I'd like a system where the ADC data is written into the RAM of the Red Pitaya at 125Msps and then later sent to my computer for processing, where I can then see the FFT of the waveform and/or save it to my computer. Another thing I would like is to be able to send a file to the Red Pitaya containing an arbitrary waveform and have the DAC play this waveform over and over again. I'm pretty sure you have already done both of these things, but I was wondering how I would get your projects to work on the STEMLab 125-10, which has half the amount of RAM as the STEMLab 125-14. I also tried another project which is based on yours (https://github.com/alex123go/RedPitaya_Acquisition), but I get bad data when I tried installing it as that other project was designed to work with 512MB of RAM.

How would I modify this project to work on the STEMLab 125-10? I think I can get away with using a small amount of RAM (like 64MB), and having the rest be used as a buffer for the ADC data. Thanks!

pavel-demin, 2022-06-30:
I think projects in this repository should work with STEMLab 125-10 without modifications.

I recently switched to using Contiguous Memory Allocator (CMA) for DMA buffers and as far as I remember current versions of projects in this repository do not have hard coded RAM addresses.

The size of memory available for CMA is controlled by the `cma` parameter in [uEnv.txt](https://github.com/pavel-demin/red-pitaya-notes/blob/master/uEnv.txt#L15).

If you want to maximize the amount of RAM available for your application, I think you should use the [SD card image based on Debian](http://pavel-demin.github.io/red-pitaya-notes/led-blinker/#sd-card-image).

EggertEnjoyer123, 2022-06-30:
Thanks for your help!

What modifications would I need to make to account for the adc_recorder project to account for the different memory layour?

pavel-demin, 2022-06-30:
I think the only modifications are to set `writer_0/ADDR_WIDTH` to 24 in [block_design.tcl](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/adc_recorder/block_design.tcl#L220) and `cma` to 132M in [uEnv.txt](https://github.com/pavel-demin/red-pitaya-notes/blob/master/uEnv.txt#L15) if you want to have a 128 MB DMA buffer.

However, before you start modifying the adc_recorder project, I recommend playing around with the writer_test project to ensure that data can be continuously written to RAM at 125 MSPS. I have had cases recently where I have observed issues at this sample rate. Maybe FIFO buffer size in axis_ram_writer should be increased.

pavel-demin, 2022-06-30:
I have just added with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/f6bcd8841223cde795cf647894b72548c0d362b1) a parameter for the FIFO buffer size to projects that use axis_ram_write. Now the writer_test project writes continuously to RAM at 125 MSPS without problems.

EggertEnjoyer123, 2022-07-01:
Can I ask where the numbers come from? (like 24 and 132M).
I am very new to embedded systems.

EggertEnjoyer123, 2022-07-01:
Also, when I run free on the Red Pitaya after installing, I get this. 
free -m
              total        used        free      shared  buff/cache   available
Mem:            497          16         440           3          40         468
Swap:             0           0           0

Do you know why it's showing 497MB even though there's only 256MB ram on the device? Whenever I try to do anything complicated (like use apt) it also crashes.

pavel-demin, 2022-07-01:
If I am not mistaken the memory size is configured in the devicetree files for U-Boot and Linux.

Editing the Linux devicetree file is easy:
- convert devicetree.dtb to devicetree.dts
```
dtc -I dtb -O dts -o devicetree.dts devicetree.dtb
```
- edit devicetree.dts and replace 0x20000000 with 0x10000000 in the memory block:
```
        memory {
                device_type = "memory";
                reg = <0x00 0x20000000>;
        };
```
- convert devicetree.dts to devicetree.dtb
```
dtc -I dts -O dtb -o devicetree.dtb devicetree.dts
```
If it does not help, then you will need to edit [patches/zynq-red-pitaya.dts](https://github.com/pavel-demin/red-pitaya-notes/blob/master/patches/zynq-red-pitaya.dts) and rebuild U-Boot and boot.bin.

EggertEnjoyer123, 2022-07-01:
How would I rebuild the U-Boot stuff from the dts file?

pavel-demin, 2022-07-01:
> How would I rebuild the U-Boot stuff from the dts file?

```
make clean
make boot.bin
```

EggertEnjoyer123, 2022-07-02:
Is there a way to make the files without installing Vitis?
I don't think I have enough disk space on my computer.

pavel-demin, 2022-07-02:
I think it is possible to build U-Boot without Vitis but I do not have a ready made recipe. You will still need Vitis/Vivado to build adc_recorder and other projects from this repository.

Here is a link to an SD card image with 0x20000000 replaced with 0x10000000 in both devicetree files:

https://www.dropbox.com/sh/5fy49wae6xwxa8a/AAChqMZoTAXjHS4PcqlXbzvBa/red-pitaya-debian-11.3-armhf-20220702.zip?dl=1

It is possible that my scripts building the FPGA bitstream files still work with the old versions of Vivado. In this case, you can try installing Vivado 2018.3 that uses much less disk space than the newer version. Here is a link to Vivado 2018.3:

https://www.xilinx.com/member/forms/download/xef-vivado.html?filename=Xilinx_Vivado_SDK_2018.3_1207_2324.tar.gz

EggertEnjoyer123, 2022-07-02:
Cool, it seems like I'm getting the right amount of memory now:
`root@red-pitaya:~# free -m
               total        used        free      shared  buff/cache   available
Mem:             241          18          75           0         147         215
Swap:              0           0           0
`

EggertEnjoyer123, 2022-07-03:
It seems like there is an error message when I try running the .tcl file.
Which version of Vivado are you using? I'm using 2020.1
```
start_gui
source /home/eggert/red-pitaya-notes/projects/adc_recorder/block_design.tcl
# cell xilinx.com:ip:clk_wiz pll_0 {
#   PRIMITIVE PLL
#   PRIM_IN_FREQ.VALUE_SRC USER
#   PRIM_IN_FREQ 125.0
#   PRIM_SOURCE Differential_clock_capable_pin
#   CLKOUT1_USED true
#   CLKOUT1_REQUESTED_OUT_FREQ 125.0
#   USE_RESET false
# } {
#   clk_in1_p adc_clk_p_i
#   clk_in1_n adc_clk_n_i
# }
invalid command name "cell"
    while executing
"cell xilinx.com:ip:clk_wiz pll_0 {
  PRIMITIVE PLL
  PRIM_IN_FREQ.VALUE_SRC USER
  PRIM_IN_FREQ 125.0
  PRIM_SOURCE Differential_clock_capable_pin
  C..."
    (file "/home/eggert/red-pitaya-notes/projects/adc_recorder/block_design.tcl" line 2)
    ```

EggertEnjoyer123, 2022-07-03:
Never mind I figured out how to get the bitstream

EggertEnjoyer123, 2022-07-03:
So I installed the new bitstream and I compiled adc-recorder.c, and it seems to spit out valid numbers.
I'm pretty sure now I have to figure out how to change your code to save the results to a file, and then send that file via Ethernet to my computer where I can do processing with Python or something. Any tips?

EggertEnjoyer123, 2022-07-04:
Ok so I took a look through adc-recorder.c
The page size is 4096, so 4096 * 1024 => 4MB of CMA memory allocated.
This makes sense, because 1 million samples * 2 channels * 2 bytes/sample gets you to 4MB too.
The problem now is that if I try allocating 32 times as much memory, I get an ioctl error, even though I should have 128MB DMA. The most I can allocate is somewhere between 21 * 1024 * 4096 and 22 * 1024*4096.  Do you know what could be causing this and how I can fix it? Thanks.

pavel-demin, 2022-07-04:
Try increasing the `cma` parameter value and check `cat /proc/meminfo | grep -i cma`. I could allocate 128MB after setting the `cma` parameter to 168M.

EggertEnjoyer123, 2022-07-04:
Do you happen to know why changing 
`  ram = mmap(NULL, 1024*sysconf(_SC_PAGESIZE), PROT_READ|PROT_WRITE, MAP_SHARED, fd, 0);`
to 
` ram = mmap(NULL, size, PROT_READ|PROT_WRITE, MAP_SHARED, fd, 0)`
causes a segfault even though size = 1024 * sysconf(_SC_PAGESIZE) ?

pavel-demin, 2022-07-04:
After the `ioctl` call `size` contains the starting address of the allocated memory.

EggertEnjoyer123, 2022-07-04:
I think you're right.
Also I changed the CMA value to a larger number and I am able to allocate more memory now and get more samples. I think I will need to send the data over to my computer so I can do an FFT on it and display the results, but I don't think that should be too hard. Thanks for your help!
Also, if I were to make modifications to the block_diagram.tcl file, how would I go about doing it? (Like lets say I wanted to add another feature). I couldn't figure out how to do it in Vivado.

EggertEnjoyer123, 2022-07-04:
Hmm i was pretty sure I had it set to 24, but it seems like something reverted that change.
Making that change again fixed the problem.
Also, do you know why sometimes ioctl reports that it failed to allocate memory (even though there is enough memory)? My guess is that the operating system is still competing for memory. Finally, do you know how I would modify the .tcl files with vivado? I'm getting XML character encoding not supported

pavel-demin, 2022-07-04:
I agree that CMA behaves strangely. I have never used more than 32MB of CMA memory before. So I am discovering these issues from your tests.

The only thing the `ioctl` call does is allocate memory using `dma_alloc_coherent`:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/patches/cma.c#L34

I do not know why `dma_alloc_coherent` sometimes fails when trying to allocate 128MB. Maybe CMA is not designed to work with more than half the total RAM.

I do not use Vivado to edit .tcl files, I do it with a simple text editor.

If you want to use Vivado to edit your project, then the following commands can be used:
- build Vivado project
```
make NAME=project_name xpr
```
 - open project in Vivado
```
vivado tmp/project_name.xpr
```
Other examples can be found in the slides at [this link](https://www.dropbox.com/sh/5fy49wae6xwxa8a/AACl--BhQvcNgjeQLRaiX9dha/ClubVivado2016_Pavel_Demin.pdf?dl=1).

closed 2022-08-07 by pavel-demin

mabl, 2022-10-31:
I applied the following patch to make the images work with the STEMLab 125-10:

```patch
diff --git a/cfg/red_pitaya.xml b/cfg/red_pitaya.xml
index c48ea199..7a3e89e9 100644
--- a/cfg/red_pitaya.xml
+++ b/cfg/red_pitaya.xml
@@ -28,7 +28,7 @@
   <set param="PCW::SPI1::SPI1::IO" value="MIO 10 .. 15" />
   <set param="PCW::PRESET::BANK0::VOLTAGE" value="LVCMOS 3.3V" />
   <set param="PCW::PRESET::BANK1::VOLTAGE" value="LVCMOS 2.5V" />
-  <set param="PCW::UIPARAM::DDR::PARTNO" value="MT41J256M16 RE-125" />
+  <set param="PCW::UIPARAM::DDR::PARTNO" value="MT41J128M16 HA-187E" />
   <set param="PCW::UIPARAM::DDR::BUS_WIDTH" value="16 Bit" />
   <set param="PCW::MIO::MIO[0]::PULLUP" value="disabled" />
   <set param="PCW::MIO::MIO[16]::SLEW" value="fast" />
diff --git a/patches/zynq-red-pitaya.dts b/patches/zynq-red-pitaya.dts
index 1b40963e..41c45396 100644
--- a/patches/zynq-red-pitaya.dts
+++ b/patches/zynq-red-pitaya.dts
@@ -18,7 +18,7 @@
 
        memory@0 {
                device_type = "memory";
-               reg = <0x0 0x20000000>;
+               reg = <0x0 0x10000000>;
        };
 
        chosen {

```

adithya-n007, 2023-07-07:
> I think it is possible to build U-Boot without Vitis but I do not have a ready made recipe. You will still need Vitis/Vivado to build adc_recorder and other projects from this repository.
>
> Here is a [...]

Hey Pavel, can you resend the SD card image for RedPitaya StemLab 125-10?. With the current image, the system crashes when I try to install gcc or build-base.

guillaume5, 2024-06-20:
For the system to boot properly, I also added mem=256M to bootargs in uEnv.txt.

Due to lack of tmpfs I add to also increase / (=filesystem root) and installed some packages with apk add.
"df -m" -> diskfree display % of use
"mount -o remount,size=200M,noatime /" -> to extend tmpfs