pavel-demin/red-pitaya-notes #1054: How to start the receiver in sdr_receiver_hpsdr_122_88 mode?

w3ua, 2022-07-08:
### Description of the setup:

 * Device: Red Pitaya 122_88 with External Clock
 * SD card image: 20220322
 * Application: SkimServer
 * Other relevant information:

### Description of the problem:

I copied start.sh file to the root directory of SD card, as prescribed. But every time I boot the receiver, it starts in app selection state, so to start using it as a skimmer server, I need manually open http://xx.xx.xx.xx (my RP ethernet interface) and select SDR Receiver HPSDR 122-88 from the menu. Only then my Skimmer servers discover the receiver.

Ideal solution would be to create the single-use distribution (with no other apps but HPSDR receiver) and make it work from the beginning -- unfortunately, my skills with embedded linux preclude me from doing that. Any help will be appreciated

### Steps to reproduce the problem:

 1. Install the SD image
 2. copy start.sh to the root
 3. boot.

davidhay45, 2022-07-09:
I think you must be copying the HPSDR start.sh file to the wrong directory.  It has to go to the /media/mmcblk0p1 directory.  It certainly works on my system.

pavel-demin, 2022-07-10:
I agree with the previous comment. Looks like start.sh was copied to the wrong directory. The file should be copied to the topmost directory of the SD card. This directory contains boot.bin, devicetree.dtb, uEnv.txt, uImage.

I had single-use distributions in the past, but with the growing number of apps, it became difficult to maintain them. So I plan to keep the current system for a while.

w3ua, 2022-07-10:
Putting file start.sh to the root directory of FAT32 filesystem in the primary partition of the SD card created it in the directory  @davidhay45 mentioned:

-----------------------------------------------------------------------------------------------
rp-f07bc5:/media/mmcblk0p1# ls
System Volume Information  cache                      red-pitaya.apkovl.tar.gz   uImage
apps                       devicetree.dtb             **start.sh**                   uInitrd
boot.bin                   modloop                    uEnv.txt                   wifi
rp-f07bc5:/media/mmcblk0p1# cat start.sh
#! /bin/sh

apps_dir=/media/mmcblk0p1/apps

source $apps_dir/stop.sh

cat $apps_dir/sdr_receiver_hpsdr_122_88/sdr_receiver_hpsdr_122_88.bit > /dev/xdevcfg

address=`awk -F : '$5="FF"' OFS=: /sys/class/net/eth0/address`

echo 2 > /proc/sys/net/ipv4/conf/all/arp_announce
echo 1 > /proc/sys/net/ipv4/conf/all/arp_ignore
echo 2 > /proc/sys/net/ipv4/conf/all/rp_filter

ip link add mvl0 link eth0 address $address type macvlan mode passthru

$apps_dir/sdr_receiver_hpsdr_122_88/sdr-receiver-hpsdr eth0 1 1 1 1 1 1 1 1 &
$apps_dir/sdr_receiver_hpsdr_122_88/sdr-receiver-hpsdr mvl0 1 1 1 1 1 1 1 1 &
rp-f07bc5:/media/mmcblk0p1#
-----------------------------------------------------------------------------------------------------------
So the start.sh is definitely there -- but it doesn't work for me. Every tome I reboot my RP, I sill need to go to the RP URL and start the receiver app manually.
What's wrong?
Thank you.

davidhay45, 2022-07-11:
Check that the start.sh has the execute attribute.  This is what my start.sh contains:

rp-f099d6:~# cat /media/mmcblk0p1/start.sh
#! /bin/sh

apps_dir=/media/mmcblk0p1/apps

source $apps_dir/stop.sh

cat $apps_dir/sdr_transceiver_hpsdr_122_88/sdr_transceiver_hpsdr_122_88.bit > /dev/xdevcfg

$apps_dir/sdr_transceiver_hpsdr_122_88/sdr-transceiver-hpsdr 1 2 2 2 1 2 &
rp-f099d6:~#

w3ua, 2022-07-14:
start.sh is definitely executable:

rp-f07bc5:/media/mmcblk0p1# ls -l
total 30158
drwxr-xr-x    2 root     root          8192 Jul  7 21:53 System Volume Information
drwxr-xr-x   29 root     root          8192 Mar 22 21:36 apps
-rwxr-xr-x    1 root     root        530336 Mar 22 19:10 boot.bin
drwxr-xr-x    2 root     root         16384 Jan  1  1980 cache
-rwxr-xr-x    1 root     root         12497 Mar 22 19:18 devicetree.dtb
-rwxr-xr-x    1 root     root      22048768 Mar 22 21:31 modloop
-rwxr-xr-x    1 root     root          9162 Mar 22 21:36 red-pitaya.apkovl.tar.gz
**-rwxr-xr-x**    1 root     root           594 Mar 22 21:36 start.sh
-rwxr-xr-x    1 root     root           506 Oct 28  2021 uEnv.txt
-rwxr-xr-x    1 root     root       5224368 Mar 22 19:18 uImage
-rwxr-xr-x    1 root     root       3008195 Mar 22 21:31 uInitrd
drwxr-xr-x    2 root     root          8192 Mar 22 21:36 wifi
rp-f07bc5:/media/mmcblk0p1#

pavel-demin, 2022-08-04:
I have just tried to reproduce the problem.

I have formatted an SD card, unpacked `red-pitaya-alpine-3.14-armv7-20220322.zip` and copied `start.sh` from `apps/sdr_receiver_hpsdr_122_88` to the topmost directory of the SD card:
```
# cp apps/sdr_receiver_hpsdr_122_88/start.sh .
# ls -l 
total 30224
drwxr-xr-x 29 root root    16384 Mar 22 21:36 apps
-rwxr-xr-x  1 root root   530336 Mar 22 19:10 boot.bin
drwxr-xr-x  2 root root    16384 Mar 22 21:36 cache
-rwxr-xr-x  1 root root    12497 Mar 22 19:18 devicetree.dtb
-rwxr-xr-x  1 root root 22048768 Mar 22 21:31 modloop
-rwxr-xr-x  1 root root     9162 Mar 22 21:36 red-pitaya.apkovl.tar.gz
-rwxr-xr-x  1 root root      594 Aug  4 12:21 start.sh
-rwxr-xr-x  1 root root      506 Oct 28  2021 uEnv.txt
-rwxr-xr-x  1 root root  5224368 Mar 22 19:18 uImage
-rwxr-xr-x  1 root root  3008195 Mar 22 21:31 uInitrd
drwxr-xr-x  2 root root    16384 Mar 22 21:36 wifi
```
When I boot Red Pitaya STEMlab 122.88-16 SDR with this SD card, the `sdr_receiver_hpsdr_122_88` application starts automatically and I see two  `sdr-receiver-hpsdr` processes running:
```
# ps aux | grep sdr
 2209 root      0:00 /media/mmcblk0p1/apps/sdr_receiver_hpsdr_122_88/sdr-receiver-hpsdr eth0 1 1 1 1 1 1 1 1
 2210 root      0:00 /media/mmcblk0p1/apps/sdr_receiver_hpsdr_122_88/sdr-receiver-hpsdr mvl0 1 1 1 1 1 1 1 1
```
So, for me, everything works without problems.

closed 2022-08-19 by pavel-demin