pavel-demin/red-pitaya-notes #1056: /dev/cma open error while running adc-test/server

glnmurthy, 2022-08-17:
### Description of the setup:

 * Device:
 * SD card image:
 * Application:
 * Other relevant information:

### Description of the problem:

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2022-08-17:
What SD card image are you using?

Normally, the latest version of the SD card image from this repository ([red-pitaya-alpine-3.14-armv7-20220322.zip](https://github.com/pavel-demin/red-pitaya-notes/releases/download/20220322/red-pitaya-alpine-3.14-armv7-20220322.zip)) contains /dev/cma.

pavel-demin, 2022-08-17:
Here is the additional information from #1057:

 * Device: redpitaya 125-10
 * SD card image: supplied by Redpitaya
 * Application: ADC Test for 125Msps

pavel-demin, 2022-08-17:
The projects in this repository that use the CMA driver and `/dev/cma` work with SD card images from this repository.

The source code of the driver is in [patches/cma.c](https://github.com/pavel-demin/red-pitaya-notes/blob/master/patches/cma.c). I think it should be possible to add this driver to the SD card image provided by Red Pitaya. However, I think that it would be much easier to use an SD card image from this repository.

Additionally, projects in this repository are only tested with STEMlab 125-14 and STEMlab 122.88-16 SDR. To work with STEMlab 125-10 they would probably have to be modified. Some information about the modifications can be found in #1052.

closed 2022-08-19 by pavel-demin

glnmurthy, 2022-10-11:
I am using the same