pavel-demin/red-pitaya-notes #1057: Unable to open cma in ADC Server .c used in adc test

glnmurthy, 2022-08-17:
### Description of the setup:

 * Device:redpitaya 125-10
 * SD card image: Supplied by Redpitaya
 * Application: ADC Test for 125Msps
 * Other relevant information:
used the code ADC Test client on Redpitaya getting error of cma access /not defined -think cma is for memory allocation and cache but it seems the board OS not supporting this driver ,the problem is same for all other adc codes that's tried for getting 125Msps without usage of cma in the code ,with cma ,the code is compiled with errors

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2022-08-17:
Thank you for this additional information.

Could you please copy this information to the issue about the same problem that you opened earlier today and close this one?

I will post my answer in the previous issue (#1056).

pavel-demin, 2022-08-17:
I think it would be easier if I copy the information myself. Closing this issue.

closed 2022-08-17 by pavel-demin