pavel-demin/red-pitaya-notes #1058: Vivado 2021.1 open project with TCL Command

FarukSozuer, 2022-08-19:
Hello;

I'm using Vivado 2021.1. I try this project in this version. But I'm not open project. When I show the project path related to the cd on the TCL screen and reference the source and the .tcl file, I get the following errors. I want to try the project by making some changes on the block design.

[image]

I couldn't fix the error. I am waiting for your support. Sincelery

pavel-demin, 2022-08-19:
Scripts in this repository are meant to be run through Makefile as shown in the examples in the notes. They work with Vitis/Vivado 2020.2. I do not use the newer versions due to some issues with the new DSP macro IP core. I posted a description of these issues on the Xilinx forum:

https://support.xilinx.com/s/question/0D52E00006iHpkX

It is possible to run the scripts directly without Makefile. Here are the instructions for running them on a Windows machine:

**a) Open Vivado, and in the Tcl console, paste this code:**
```
cd c:/#your-path-to-the-code#/red-pitaya-notes-master
source helpers/build-cores.tcl
```

This will generate the cores in the tmp directory.

**b) Build the project**
```
set argv [list sdr_receiver xc7z010clg400-1]
source scripts/project.tcl
```

FarukSozuer, 2022-08-19:
Thanks for reply. Problem is solved. 

Sincelery

closed 2022-08-19 by pavel-demin

ByoungjinKim, 2022-12-27:
invalid command name "cell"

oh, good solution

thanks