pavel-demin/red-pitaya-notes #1059: Stemlab 128.88-16 Origin of TX spurs in 3rd Nyquist Zone and possible remedy?

hb9ewy, 2022-08-19:
### Description of the setup:

 * Device: Stemlab 122.88-16, V1R1
 * SD card image: red-pitaya-alpine-3.14-armv7-20220322.zip
 * Application: "SDR transceiver compatible with HPSDR"
 * Other relevant information: SDR Quisk 4.1.96, but same behavior with powerSDRmx 
 
### Description of the problem:
- Spurs on TX in 2m Band close to the carrier (e.g. about 192kHz), only 50..60dBc, depenig on the selected I/Q data rate (rx sampling rate on quisk).
- The strength of the spurs depends on the transmitted carrier level in a linear manner.
- see SA screenshots below: 
[image]
[image]

### Steps to reproduce the problem:

 1. tx a constant carrier in the 2m band. e.g. fundamental about 21.41.46MHz -> 144.34MHz in 3rd NQZ, level below fs.
 2. observe an measure surious emissions close to carrier
 3. vary the I/Q data rate: 48, 96, 192, 384 kHz. 

I wonder what causes these spurs and why do they vary with the I/Q data rate (I assume this to be the rate of the the rx data stream).
Is my assumption correct, that the tx I/Q data rate is always 48kHz?

Thank you.

pavel-demin, 2022-08-19:
Thank you for the tests. I will try to reproduce this effect.

> Is my assumption correct, that the tx I/Q data rate is always 48kHz?

Yes, the TX I/Q rate is fixed and it is always 48k.

hb9ewy, 2022-08-20:
Thanks Pavel.
With the TX I/Q rate always being 48 kHz I really wonder what makes the observed spurs to depend on the RX I/Q rate.
May be crosstalk in the processing chain of the DAC clock in the HW or FPGA? Measurments of the signal are difficult, but I may try.

BTW, the fundamental signal has these spurs at better than -80dBc, measured at 10.7 MHz with a notch filter supressing the carrier.
Yesterday night I measured TX in the 3rd NQZ on a modified Red Pitaya 14 Bit - spurs are stronger and there are more. The 122.88-16 is definitely cleaner.

For my application I consider to take the fundamental tx signal and upconvert it by the 122.88 MHz clock (using an older transverter which used an external 130MHZ LO).

hb9ewy, 2022-08-20:
Hi Pavel,
I had been in contact with DC9OE, and he experienced similar spurs, caused probably by an on board SMPS.
With an H-Field probe I found considerable 200kHz fields on the PCB; and also on 1.8V lines. Additional filtering on the 3V3A supply of the DAC did not improved matters.
So SMPS noise is probaly an important contributor of these spurs. Though, I still wonder what is the cause of the dependency with the RX I/Q rate.

From my point of view this issue can be closed.

pavel-demin, 2022-08-23:
Thank you for testing different versions.

Looking at the changes between 20200628 and 20210427, I think only the following changes could affect this SDR application:

- added fifth receiver to SDR transceiver compatible with HPSDR for better compatibility with Thetis (https://github.com/pavel-demin/red-pitaya-notes/commit/907db3b37734206904b29c19e82d2e1ba58d08fa, https://github.com/pavel-demin/red-pitaya-notes/commit/d61ef4f6a5b82f3304256a3e5b4812231783e6be)
- improved DAC interface and PLL configuration (https://github.com/pavel-demin/red-pitaya-notes/commit/40ed371c574e673c482afd147b5c3712e187f985, https://github.com/pavel-demin/red-pitaya-notes/commit/20e3d8007c3314246837afed689b6ea3692c9d5e)

hb9ewy, 2022-08-23:
Hi Pavel,
I looked already at these changes, but have too few knowledge about FPGA programming to understand them. 
For now I will use the oldest 3.12 FW and reduce the power supply voltage to reduce the SMPS spurs.
The main issue for tx is high DAC clock noise. In the 3rd NQZ naturally it gets worse, I only measured about -110dBc at +/- 100kHz. I need at least 20dB better for 144MHz. NO chance with this HW. I hope the future releases of the HW will improve.
Once HW has improved (which I hope it will) these spurs may need attention again.

Thanks for looking into this issue. 
For me it could be moved into the backlog or even closed.