pavel-demin/red-pitaya-notes #1060: Corrupted sine wave with Wideband SDR Transceiver

dxxx9, 2022-08-22:
### Description of the setup:

 * Device: Red Pitaya 125-14
 * SD card image: your Alpine image
 * Application: Wideband SDR Transceiver

### Description of the problem:

Hello Pavel,

I looked at #930 because I'm also interested in baseband transmission. My flowgraph is similar to [yours](https://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-wide/). OUT1 is connected with IN1. But the received sine wave doesn't look like expected at all. This happens for the higher selectable sample rates (most severe at 2500 kSPS). See image below.

[image]

There are arbitrary phase shifts and parts where there is no signal at all.

As I'm not sure how to find the source of error, I'm writing this issue.
Can this be caused by the network connection?

pavel-demin, 2022-08-22:
The "Sample Rate" fields in the Red Pitaya blocks of your diagrams are empty. So maybe these blocks do not run at the same sample rate as the other two blocks.

dxxx9, 2022-08-23:
Thanks for your reply. The sample rate is not the issue. See the images below for 0.5MSPS, 1.25MSPS and 2.5MSPS.

0.5MSPS works fine 100% of the time.
[image]

1.25MSPS results are as expected 50% of the time.

https://user-images.githubusercontent.com/62398752/186126096-8b76a114-bded-474e-b6f5-00e64e6b0e40.mov

2.5MSPS has those gaps and arbitrary phase shifts all the time.
[image]

What I noticed with the 2.5MSPS rate, is that the pattern of the phase shifts (or the new waveform) is almost the same for multiple periods, until the next gap. Then a new pattern starts.
[image]

- Do you have any clue how to narrow down the error?
- Could this be caused by network issues?
- Can you reproduce this issue when choosing the upper sample rates?

pavel-demin, 2022-08-23:
Thank you for the tests.

I think I have not tested 2.5 MSPS before because I did not have a fast Linux computer to run GNU Radio at that sample rate.

I just did some tests:

 - 1.25 MSPS rate works without problem
[image]

 -  2.5 MSPS rate does not work well
[image]

The CPU load on the Red Pitaya board is quite high. It seems that setting the RX and TX sample rate to 2.5 MSPS does not work for this application.

dxxx9, 2022-08-23:
Ok, thanks for your time and confirming this issue.

Interesting that your 1.25MSPS signal looks fine. Did you use the `RP 125-14` as well?

pavel-demin, 2022-08-23:
Yes, I used Red Pitaya 125-14 for the tests. I connected the Ethernet ports of the Red Pitaya board and my computer directly with a good CAT6 cable.

dxxx9, 2022-09-16:
Thank you Pavel, for the clarification.
If I understand correctly, you are assuming that the CPU is the limiting factor in this case?

I am transmitting PAM in baseband and that is why I have set the carrier frequency to zero.
Is it possible to increase the receive bandwidth further?
It seems to me that the board is designed for radio applications that use carrier modulation. The fast ADC allows passband transmission up to 60 MHz, but without DDS the CPU rate is not sufficient to generate baseband signals with the same frequency (about 60 MHz). Correct?

Greetings, Daniel

pavel-demin, 2022-09-18:
Most of the projects in this repository aim for simplicity rather than high performance to show that fairly simple techniques can be used to build some interesting applications for the Red Pitaya board. It should be possible to alleviate some bottlenecks using more complicated techniques.

> Is it possible to increase the receive bandwidth further?

The [sdr_receiver_wide_122_88](https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/sdr_receiver_wide_122_88) project has two RX channels with a maximum sample rate of 10.24 MSPS per channel. I think that it should be possible to modify it to have one RX channel with a maximum sample rate of 15.36 MSPS.

dxxx9, 2022-10-16:
Alright, thanks again!

closed 2022-10-16 by dxxx9

pavel-demin, 2023-08-10:
The better performing version the wideband SDR transceiver is now in the [new release](https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20230705).