pavel-demin/red-pitaya-notes #1061: PowerSDR/Pitaya connection failure when selecting between FM/SSB modes

JoNoci, 2022-08-26:
### Description of the setup:

* PC:  Win10 3.4GHz 16GB ram 4cores
* Device: RED Pitaya
  * SD card image: red-pitaya-alpine-3.14-armv7-20220322
 * Application:   PowerSDR Charly 25 HAMlab STEMlab Edition
        As downloaded from the click-link on PAVEL NOTES site
 * Pitaya obtains IP via DHCP. Pitaya is connected to DHCP router, PC is connected to same router, also receives IP via DHCP
 

 **Description of the problem:**
Run PowerSDR, click on 'Power' and SDR runs fine - Pitaya is found and everything works fine in SSB/AM and like modes
However, selecting FM results in a definite Audio 'glitch', PowerSDR halts and the 'Error starting HPSDR Hardware..' screen appears.
At this point all is dead - Power button causes same screen popup. Re-start PowerSDR - CHANGE THE MODE TO DESIRED MODE FIRST - then click on Power and it works. If FM was selected before Power activated, it works, but click on SSB and the same failure occurs.

[image]

pavel-demin, 2022-08-26:
This seems to me to be a software problem. PowerSDR Charly 25 HAMlab STEMlab Edition may require a special version of the application running on the Red Pitaya board and may not be compatible with the red-pitaya-alpine-3.14-armv7-20220322 SD card image.

Please try PowerSDR mRX PS or Thetis from the following links:

https://github.com/TAPR/OpenHPSDR-PowerSDR/releases

https://github.com/DH1KLM/Thetis-2.9.0-Red-Pitaya-DH1KLM/releases

JoNoci, 2022-08-26:
Thank You Pavel. I installed the mRX version you linked to, and it works perfectly.
Much appreciated! 
Thanks for all your very good work.

closed 2022-08-26 by JoNoci