pavel-demin/red-pitaya-notes #1062: Activation of discrete outputs DIO-1/2/3 for attenuator/preamp on Red Pitaya with PowerSDR

JoNoci, 2022-08-31:
### Description of the setup:

 * Red Pitaya 122-16
 * red-pitaya-alpine-3.14-armv7-20220322
 * PowerSDR
  Intention is to build own Receive preselector filters and TX amp filters, controlled by Pitaya as HERMES Control

### Description of the problem:
Setup is working fine, but lacking information/Docs on Pitaya discretes I/O operation with Powersdr.
In PowerSDR setup hardware config is HERMES (not HPSDR - if set to HPSDR, errors out when POWER button is clicked, says METIS firmware is incorrect version, etc).
Under H/W options, Alex/Apollo are not selected.
Under Hermes Ctrl button, I set a binary sequence in the tick boxes in J16 RECEIVE PINS - 
160m = 0001
80m = 0010
60m = 0011, etc
Then when I change bands in PwrSDR, the 4 HERMES Control discrete outputs on the Pitaya change to the set values in that list. This enables a 4 to 16 line decoder to do filter selection for RX/TX, etc. This all works fine.

But how do I get the 3 attenuator/preamp enable discretes to work via PwrSDR?
You state :
_**output for a pre-amplifier/attenuator control is connected to pin DIO1_P of the [extension connector E1]) (this pin is controlled by the first ATT combo-box in [PowerSDR mRX PS]
outputs for 10 dB and 20 dB attenuators control are connected to the pins DIO2_P - DIO3_P of the [extension connector E1]**_

but I never get these discretes to activate. Where is the  **_first ATT combo-box _** and how do I make it control those discretes and/or how do I get to control/activate those discretes from within PwrSDR?
Using the ATT selector ( bottom left on PwrSDR, under the DRIVE slider) does not activate the discretes, and I have nor found a pre-amp on/off control...

I have searched the internet, OpenHPSDR, etc...
Thank You.

pavel-demin, 2022-09-01:
> Under H/W options, Alex/Apollo are not selected.

The following signals are connected to DIO1_P-DIO3_P:
 - Preamp On/Off (0 = Off, 1 = On)
 - Alex Attenuator (00 = 0dB, 01 = 10dB, 10 = 20dB, 11 = 30dB)
 
I think that for ATT signals to work, Alex needs to be enabled.

pavel-demin, 2022-09-01:
The Alex interface is active if no external audio codec is connected. So maybe connecting a logic analyzer to the Alex interface would be the easiest way to see how PowerSDR controls Alex.

For the I2C interface to work, it must be connected to a supported I2C module. For example, audio codec or [PCA9555](https://www.mouser.com/c/?q=PCA9555) or [NUCLEO-G071RB](https://www.mouser.com/ProductDetail/511-NUCLEO-G071RB) with [this firmware](https://github.com/pavel-demin/nucleo-g071rb-examples/tree/master/i2c_to_spi).

JoNoci, 2022-09-01:
I realise there must be a recipient interface on I2c, such as PCA9555, for anything real to happen beyond the Pitaya I/O - that is not the issue. If we assume that a PCA9555 is connected, what must be enabled/ticked in PwrSDR setup for the PwrSDR/Pitaya combo to talk to that 9555? IAW, if the 9555 is part of ALEX/HERMES or whatever and selects filters or something, what must I tick in setup, and what must I do in PwrSDR to cause an I2C communication to take place?
The same applies for the SPI bus.
I have connected a logic analyser on all the I/O lines ( SPI, I2C, 4xHERMES Ctrl, TX/RX strobes, UART-TX/RX, , etc) and am trying to understand hat generates an I/O on those pins from PwrSDR. You suggested I need to select ALEX if I want the ATT function to work, but the result is nonsensical on the DIO-1/2/3 pins...
And no matter what I do in PweSDR ( change frequencies, bands, ATT, Swap modes, etc, there is never I/O on I2C or SPI. 
There is I/O on TX/RX Strobe out, Clock Out, serial data out, which appears to be for serial shift registers, and the data clocked out is consistent and makes sense per band, etc.
There is I/O on the Uart TX line, but not yet making sense per band etc.

Also, When I start PwrSDR, if in setup I select HPSDR and click POWER in PwrSDR, it errors out with a statement of incorrect firmware, metis, mercury, etc.
If I tick 'ignore firmware revision', it gets past. I have to do this to be able to select HPSDR, else I cannot tick the MERCURY.PENNYLANE  box to try get SPI/I2C I/O to happen - so very confusing as to what I must tick and what not...

Must admit that there is no science to what I am doing, which is not my norm, just hit and miss to see what happens, and mostly nothing happens...
I have spent quite a few hours researching now, but anything related to PwSDR, the OpenHPSDR projects( Mecury, Alex, Pennylane, etc) is so old and out of date ( 2015 last update, PwrSDR updates 5 years ago, etc).

I am hoping, as a final, last resort, that a deep dig in the Pitaya source will show me what would cause an I/O function to occur , and I suppose an equally difficult dig in PwrSDR source, to see what will request such an I/O function execution.

I have two 16bit Pitayas that I used for something else, and no further use for them now - was about to give them away, and being a Ham I thought to play with SDR, but I'm not sure this is worth the difficult path! The Pitaya and your software works very well as a system - I just wanted to try and make the 'system' drive the filter relays, Preamp/Attenuator, etc via Pitaya I/O, on DIY filter hardware.

Thanks Pavel.
Regards
Joe

pavel-demin, 2022-09-01:
As far as I remember, most of the basic functionality works out of the box.

> If we assume that a PCA9555 is connected, what must be enabled/ticked in PwrSDR setup for the PwrSDR/Pitaya combo to talk to that 9555?

Nothing special must be enabled in the setup.

> IAW, if the 9555 is part of ALEX/HERMES or whatever and selects filters or something, what must I tick in setup, and what must I do in PwrSDR to cause an I2C communication to take place?

The I2C communication happens when something changes. For example, PTT is pressed, different band is selected, etc.

ka6s, 2022-09-01:
> Thanks Pavel. I will have to dig a lot more. It is a little frustrating -
> I cannot find where to look for PwrSDR operational descriptions - the
> 'official' sites have wiki's that give installation and basic setup, but
> the detail is lacking.
> I am trying to understand what PwrSDR button or function or setup
> activates which discrete I/O on the pitaya, and what functions generate I/O
> on the SPI and I2C I/O lines. I have tried to dig in the various manuals
> and build docs of ALEX and the HERMES module to understand what the
> interfaces are, but that is not easily translatable to the Pitaya I/O.
> Short of digging deep in the Pitaya and HERMES/ALEX, etc, source, I don't
> know how to find this out.
>
> In short, where can I find a description of what needs to be setup in
> PwrSDR, in HERMES config ( maybe with ALEX) , with Alpine 3.14, in order
> for the various Pitaya discretes to be activated - DIO, I2C, SPI, etc, and
> what PwrSDR 'buttons', modes, causes the I/O to occur
>
> The ALEX attenuator issue : I ticked ALEX in setup, but PwrSDR does not
> behave...The ATT button now allows these selections, in order : 0db, -20db,
> -10db, -20db, -30db, -40db, -50db.......No typo, the sequence is as I
> indicate.
> The two ATT discretes DIO2/3 do change, but not in a sensible binary
> sequence, probably since the PwrSDR ATT button sequence is nonsense.
>
> There is also no way to enable/disable only the preamp discrete - that
> seems to go inactive when any selection other than 0db is selected.
> PwrSDR is not your problem, so I need to dig elsewhere...
>
> Joe

Joe,

It is suggested now that you switch to Thetis!  Thetis is supported - AND
now can speak protocol 1 which is what Red Pitaya requires.  As an FYI - I
built my own I2C connectivity because I implemented an I2C to SPI (Alex
protocol) converter. That meant I had to slightly modify Pavel's code with
a different I2C command set (Pavel - I really wanted the same info in the
same order as SPI - so just sent THOSE over I2C instead)  It works great!
Steve KA6S

JoNoci, 2022-09-01:
Pavel,
This forum is great. Most forums I attend don't have this kind of response time - You are most diligent Pavel!
Pavel, what you said is what I assumed at the start - select ALEX and the I/O must be driven accordingly; likewise HERMES, Pennylane, Mercury, but no matter what I do in PwrSDR ( change frequencies, bands, ATT, Swap modes, TX/RX, etc, there is never I/O on I2C or SPI. - I never get the I2C or SPI to send data no matter what HPSDR hardware selections are made in setup.
 My second Pitaya does the same so I don't think the hardware is broken.

Steve:
As it is I am flabbergasted by all these strange names for the hardware - what's wrong with calling a low-pass filter a low-pass filter ( Alexiares, Mercury, penelope, etc !!) - Does not do much for clarity when one is trying to debug I/O and you have to refer  to a SpreadSheet to tell you what name belongs to what function on what boards! And then Thetis?!! What on earth is Thetis?
Is it hardware? Software? Runs on the PC or the Pitaya?  I have not managed to get the original baseline working and oYu want me to change? You really mean to punish me.

I will persevere Pavel!

Thank You

pavel-demin, 2022-09-01:
> I never get the I2C or SPI to send data no matter what HPSDR hardware selections are made in setup.

SPI is not used in this application. It is therefore normal that the SPI interface does not send any data.

I2C is only used when an I2C module with an address known to the application is connected. If you do not have a PCA9555 or a micro-controller with compatible firmware connected to the I2C bus, then the I2C interface does not send any data.

Only the Alex interface should show some activity. This Alex interface uses the pins DIO4_N (Serial Data), DIO5_N (Clock), DIO6_N (RX Strobe) and DIO7_N (TX Strobe) of the [extension connector E1](http://redpitaya.readthedocs.io/en/latest/developerGuide/hardware/125-14/extent.html#extension-connector-e1).

pavel-demin, 2022-09-01:
> What on earth is Thetis?

Thetis is a newer version of PowerSDR mRX PS.

Here is a link to a version of Thetis with some changes by DH1KLM that make it more compatible with Red Pitaya:

https://github.com/DH1KLM/Thetis-2.9.0-Red-Pitaya-DH1KLM/releases

DH1KLM, 2022-09-01:
I don't think "my version" changes anything in this regard.
From what I've read about PCA9555 in combination with Red Pitaya and PowerSDR/Thetis, it should work without any problems.

I think Yevgeni-RZ3QS is the man to ask.

[image]

JoNoci, 2022-09-01:
Thanks for the info Pavel.
WRT to the lack of SPI I/O - I believe some of my confusion  stems from perhaps misleading documentation. I have read the HPSDR ALEX manual a few times now and they say clearly the TPIC6B595 ( serial to parr chips) operate on the SPI interface...**What they meant to say** is the pins DIO4_N (Serial Data), DIO5_N (Clock), DIO6_N (RX Strobe) and DIO7_N (TX Strobe) of the [extension connector E1] - as you indicated in your response to me, and NOT SPI...So no I/O on SPI is correct.

However, WRT the I2C bus, I am still unsure of your statements. I presume the Pitaya would be I2C master or originator. In which case, if the associated hardware block is ticked in PwrSDR setup, then the Pitaya should send I2C commands on that bus with band change, etc, regardless as to weather there is an I2C peripheral attached - the Transaction should be initiated and perhaps just fail due to no response from the addressee.  The Pitaya would have to interrogate the I2C address to see what is attached, but I don't see that transaction take place. Or does it take place only on startup and if no response then never talk again?

  I suspect I have not ticked the correct box to make that transaction occur. 

What hardware should be selected in order for the I2C transactions to be required and to take place? IAW, if I just solder a PCA9555 I2C pins onto the I2C lines of the Pitaya, what hardware must be enabled for the bus to communicate? - Tick these - (?)
HPSDR?
Mercury?
Penelope?
Pennylane?
Alex?

I will try out the Thetis release - I presume the Alpine 3.14 for Pitaya is appropriate for use with Thetis...
Thanks again for your time taken to help me Pavel.
Joe

ka6s, 2022-09-01:
Jo Noci (do I have the name right).
I would like to offer you a membership in the Red Pitaya TX/RX Facebook
group. I'm one of the admins. There are lots of people on that channel that
have built Homebrew radios based around Pavel's work. (Thanks Pavel by the
way!!)  Some of us have even contributed a bit to the firmware.  Anyway - I
likely can help you with many of your questions and we don't have to take
up Pavel's time.

If your interested - just apply to join.

Also - all the interesting names come from the original HPSDR project - see
Openhpsdr.org.

73s Steve KA6S

JoNoci, 2022-09-02:
Pavel,
Thanks again for your time Pavel. I think I put this one to rest for now. Thetis and PowerSDR both seems to not control the ATT selections correctly - PwrSDr under HPSDR and HERMES generates odd sequence I/O on DIO-1/2/3 pins...And Thetis likewise odd sequences..but different from PowerSDR!

I think I will just use a manual Pre-amp/Att control to manage my hardware, like on a normal radio!

I2C issues etc I will just ignore - the 4 Hermes control line work correctly, the Serial I/O to Alex works fine, so that is good enough to decode in my own hardware to control filters, etc.  Enough for me to make it all work OK.

I will close this on the next round if thats OK...

Thanks Pavel.

pavel-demin, 2022-09-02:
> Or does it take place only on startup and if no response then never talk again?

Yes, this is exactly how it works.

> Thetis and PowerSDR both seems to not control the ATT selections correctly - PwrSDr under HPSDR and HERMES generates odd sequence I/O on DIO-1/2/3 pins...

I have just managed to install Thetis on my computer running Linux. So I can now check the ATT and Preamp signals.

The HERMES+Alex combination works exactly as it should. I do not see any problem.

Here is a table showing all the ATT settings and corresponding Preamp and Attenuator outputs:

| ATT | +20 dB Preamp (DIO1_P) | -10 dB Attenuator  (DIO2_P) | -20 dB Attenuator (DIO3_P) |
| --- | --- | --- | --- |
| 0 | On | Off | Off |
| -20 | Off | Off | Off |
| -10 | On | On | Off |
| -20 | On | Off | On |
| -30 | On | On | On |
| -40 | Off | Off | On |
| -50 | Off | On | On |

JoNoci, 2022-09-02:
Pavel, the problem really stems from a lack of basic information that can be easily accessed.
I only found the reference of the I2C bus being spoken on only on startup by digging in past github queries, some 3 years old.
Protocol documents are clear in what the data interchange is, but not WRT what occurs when, and what hardware is required for it to occur.
The same applies with PowerSDR - I have nowhere found a document that guides one through all the setup parameters for it - there are wiki's that gloss over 80% of the stuff unclear to those who do not know...There is a 'quick-setup- help doc on Open HPSDR that is really quite worthless - most of is is intuitive, and the real issues are not mentioned at all.
Likewise with Thetis - The name of the Software says Thetis --------RED PITAYA, so when you run its setup, you naturally select Radio Model = Red Pitaya, cause that's what I have. 
Then the ATT button only show 0, -10, -20 and -30dB, NO -40/-50dB....And the discretes do not change, only Bit0 changes lo if not on 0dB. 
I have searched in the associated docs - nowhere is it said I should select HERMES....But your comment :

> The HERMES+Alex combination works exactly as it should.

was the key - I set it to HERMES and it works as you say - but if you did not chance to mention HERMES, I would still be lost....
Non of this is your problem - Thetis/PwrSDR, etc is not your problem so I am not complaining At you! More sort of complaining to you to defend my stupidity a little!
It is just very difficult to find clear info on the deeper issues - the info is spread far and wide, often residing in a few key people only, with some of the pertinent web sites data becoming old and invalid...
Pavel, thanks again for your patience and your really good softwares.

I will fumble along some more!

A note to Steve - KA6S.
Thanks for the invite - I did join, but I don't really like Facebook at all - I don't find it good for technical discussion - the subject matter is lost down the page within a few hours and drifts away , lost in a messy mix of stuff....

JoNoci, 2022-09-04:
Pavel, 
I have been testing a lot of the I/O with Thetis - some issues with Thetis when the Pitaya hardware is selected, things are not consistent, but with HERMES mode the Pitaya works as advertised.
Thanks again for your patience and assistance.

Regards
Joe

closed 2022-09-04 by JoNoci