pavel-demin/red-pitaya-notes #1063: Wideband SDR support for STEMlab 122.88-16 SDR

svhum, 2022-09-15:
### Description of the setup:

 * Device: Red Pitaya STEMlab 122.88-16 SDR
 * SD card image: red-pitaya-alpine-3.14-armv7-20220322.zip
 * Application: SDR transceiver
 * Other relevant information:

### Description of the problem:

Is it possible to port the wideband SDR available for the STEMlab 125-14 to the STEMlab 122.88-16 SDR?

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2022-09-15:
I think that it is possible but I see no use for this port. There is an open issue (#1060) from a few weeks ago showing that the high sample rate does not work in the wideband SDR transceiver.

svhum, 2022-09-20:
> You don't often get email from ***@***.******@***.***>. Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

Dear Pavel,

Thank you very much for your reply. I will try to make do with the standard SDR transceiver.

I did notice an issue that the RP support teams suggested I contact you about after I first reported it to them. I am able to run the "sdr_transceiver_122_88" from the Alpine Linux image you provide on github just fine, using both GNUradio and HDSDR, on my STEMlab 122-88 SDR device. However, I tried using the version on the "Application Marketplace" available on Red Pitaya OS 1.04-11, the latest stable release I believe. I found that the SDR Tranceiver available in the Application Marketplace does not function correctly after clicking "Run". Running a sample Python script generated by GNUradio, I get the following message:

Traceback (most recent call last):
  File "/home/svhum/Utoronto/RadioAstro/red-pitaya-notes/projects/sdr_transceiver_122_88/gnuradio/trx_ssb.py", line 215, in <module>
    main()
  File "/home/svhum/Utoronto/RadioAstro/red-pitaya-notes/projects/sdr_transceiver_122_88/gnuradio/trx_ssb.py", line 193, in main
    tb = top_block_cls()
  File "/home/svhum/Utoronto/RadioAstro/red-pitaya-notes/projects/sdr_transceiver_122_88/gnuradio/trx_ssb.py", line 93, in __init__
    self.red_pitaya_source_0 = red_pitaya.source(str(addr), 1001, rx_freq, rx_samp_rate, 0)
  File "/home/svhum/Utoronto/RadioAstro/red-pitaya-notes/projects/sdr_transceiver_122_88/gnuradio/red_pitaya.py", line 39, in __init__
    self.ctrl_sock.connect((addr, port))
ConnectionRefusedError: [Errno 111] Connection refused

>>> Done (return code 1)

I also tried using HDSDR and it also does not function properly with the Marketplace version. The same script works fine with your Alpine Linux image.

Given GNUradio and HDSDR work perfectly from the Alpine Linux image, I suspect there is a small problem with the version on the Marketplace. I did notice on your Alpine Linux version that I have to refresh the home page on the Red Pitaya a few times to activate the SDR, otherwise I get the same connection refused error above (but maybe that is some cache problem).

I am just letting you know. It would be nice to have the Marketplace version working, as I prefer using the flavour of Linux Red Pitaya uses on the RP, rather than Alpine Linux, mainly because I need additional packages not available in the Alpine Linux repository needed for my end use case for the Red Pitaya. If this can be corrected somehow, please let me know!

Best regards,
Sean

pavel-demin, 2022-09-20:
The applications in the application marketplace only work with Red Pitaya STEMlab 125-14.

Putting applications to the application marketplace is too much work for me.

It should be possible to copy the `sdr_transceiver_122_88` directory from my SD card image to your preferred system, recompile the C code using `make` and modify the `apps_dir` variable in the `start.sh` script.

svhum, 2022-09-20:
Hello Pavel,

Thank you for your reply and sorry for the trouble.

I did try as you asked by compiling the transceiver directly on the Red Pitaya itself (Red Pitaya OS). Here is the result:

***@***.***:~# make
gcc -O3 -march=armv7-a -mtune=cortex-a9 -mfpu=neon -mfloat-abi=hard -o sdr-transceiver sdr-transceiver.c -lm -lpthread
***@***.***:~# ./sdr-transceiver 1
Bus error

I am not sure if you have any further ideas. I thank you again for your time.

Thanks,
Sean

pavel-demin, 2022-09-20:
Looks like the FPGA is not configured correctly. The `sdr_transceiver.bit` file must be loaded to the FPGA before running the `sdr-transceiver` program. This is exactly what the `start.sh` script does.

pavel-demin, 2022-09-21:
I have just noticed that the bitstream filename in my previous comment is not correct. The correct filename is `sdr_transceiver_122_88.bit`. Other than that, I do not see any reason why the `sdr_transceiver_122_88` application would not work with the SD card image provided by Red Pitaya developers.

svhum, 2022-09-21:
Hi Pavel,

That did the trick. I was using the .bit file provided in Marketplace distribution, which is the sdr_transceiver.bit one unsuitable for the 122_88 device.

Thank you very much for your support. I'm really glad this works now, as your transceiver will be an integral part of a project we are doing moving forward.

Regards,
Sean

closed 2022-09-21 by pavel-demin