pavel-demin/red-pitaya-notes #1064: WSPR Trasmit?

pmaine, 2022-09-18:
Thank you so much for your work!
Can you provide more details on how to setup the Red Pitaya for WSPR transmit? In your documentation you state that transmit is disabled by default.I have modified transmit-wspr-message.cfg.

How do I start transmit? Can I transmit on multiple frequencies at the same time?

Thank you!

pavel-demin, 2022-09-18:
To activate the transmit function, you need to uncomment the second line in wspr.cron. The current version transmits on a single frequency at a time.

pmaine, 2022-09-20:
I know am now able to transmit. How do I know? I unplug the antenna from my ICOM 7300 and tune to my transmit frequency of 14.095600. WSPR is being transmitted because I see it on my radio waterfall. The Red Pitya is just a few feet away from my 7300. The problem is I am not being heard when I connect my Red Pitaya to my long wire antenna. I am using a six character Maidenhead grid locator. Should I be using only four characters for Maidenhead grid locator?

Lastly, do you have a means so I can send some money to you? I really appreciate all that you have contributed!

pavel-demin, 2022-09-20:
> Should I be using only four characters for Maidenhead grid locator?

Yes, only four characters. The format should be the same as in the [example](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wspr/app/transmit-wspr-message.cfg#L15). It is the "Type 1" format. A detailed description of this format can be found at [this link](https://dxplorer.net/wspr/msgtypes.html).

> Lastly, do you have a means so I can send some money to you?

Thank you for offering. If it is OK for you I have a [paypal.me link](https://paypal.me/stemlabsdrnotes).

pmaine, 2022-09-21:
Hi Pavel
My Red Pitaya is now transmitting but I am not able to decode using WSJTX.
I have an Icom 7300 connected to a Rasberry Pi. The 7300 does not have an antenna attached. The Red Pitaya is connected to an antenna set for 20m. I am running WSJTX on the Pi. When the Red Pitya transmits - I see the transmission in the waterfall of wsjtx but wsjtx is unable to decode. I do not have GPS connected to the Pitya yet.

Is there a way to turn on log files to better understand why this is not working?

cron is configured as follows:
#1-59/2 * * * * cd /dev/shm && /media/mmcblk0p1/apps/sdr_transceiver_wspr/decode-wspr.sh >> decode-wspr.log 2>&1 &
1-51/10 * * * * cd /dev/shm && /media/mmcblk0p1/apps/sdr_transceiver_wspr/transmit-wspr.sh >> transmit-wspr.log 2>&1 &
1-59/2 * * * * cd /dev/shm && /media/mmcblk0p1/apps/common_tools/update-corr.sh 125 >> update-corr.log 2>&1 &

wspr config is:

// frequency
freq = 14.095600;

// frequency correction, from -100.0 ppm to +100.0 ppm
corr = 0.0;

// RF output channel (1 for OUT1 or 2 for OUT2)
chan = 1;

// RF output level, from -90.0 dB to 0.0 dB
level = 0.0;

// message containing station's callsign, Maidenhead grid locator and
// transmitter power in dBm
//message = "K1ABC FN42 33";
message = "KI5MIV EM10 10";

Does Alpine use NTP to sync time?

What am I doing wrong?

Thank you,
Paul

pavel-demin, 2022-09-21:
Yes, NTP is used for time synchronization. GPS is not necessary.

The transmit frequency does not look good. I think it is outside the WSPR band (14.097000-14.097200). Here is a link to a list of all the bands: https://www.wsprnet.org/drupal/node/7352

pmaine, 2022-09-22:
I am still not able to decode what I am transmitting from the Pitaya.

I changed the transmit frequency in the  transmit-wspr-message file as shown below:
// frequency
freq = 14.097100;

// frequency correction, from -100.0 ppm to +100.0 ppm
corr = 0.0;

// RF output channel (1 for OUT1 or 2 for OUT2)
chan = 1;

// RF output level, from -90.0 dB to 0.0 dB
level = 0.0;

// message containing station's callsign, Maidenhead grid locator and
// transmitter power in dBm
//message = "K1ABC FN42 33";
message = "KI5MIV EM10 10";

Here is the wspr.cron:
#1-59/2 * * * * cd /dev/shm && /media/mmcblk0p1/apps/sdr_transceiver_wspr/decode-wspr.sh >> decode-wspr.log 2>&1 &
1-51/10 * * * * cd /dev/shm && /media/mmcblk0p1/apps/sdr_transceiver_wspr/transmit-wspr.sh >> transmit-wspr.log 2>&1 &
1-59/2 * * * * cd /dev/shm && /media/mmcblk0p1/apps/common_tools/update-corr.sh 125 >> update-corr.log 2>&1 &
 
Do you have any other suggestions?

pmaine, 2022-09-22:
[image]

Here is my WSJTX while the RP is transmitting.

pmaine, 2022-09-23:
If I ssh into the RP and execute the date command I get a date time stamp from last year.

I did a ps aux and did not see NTP running.

please advise

pavel-demin, 2022-09-23:
The NTP process is called `chronyd`. If the date is from last year, then it looks like your Red Pitaya board is not connected to the internet and cannot access public NTP servers.

pavel-demin, 2022-09-23:
I see your call in the wsprnet database. Looks like your Red Pitaya board can transmit now.

pmaine, 2022-09-23:
Yes! I am transmitting on a long wire antenna using an SG230 antenna coupler. A station heard me that was 2396 km or 1488 mi from me!!! I will continue transmitting today. The grey line is approaching so I can see if I can get more distance. The coupler has about 1.1dB insertion loss and various connectors that are also introducing loss. My best estimate is I am actually transmitting between 8 and 9 dBm. I did find a small bug if I specified a dBm value less than 10 dBm in the cfg file a different number was received. For example I specified 9 dBm, 7 dBm was decoded.

I am super happy with your work!!! I want to tell others! I will update the transmit documentation and include my transmit picture and submit a pull request.

pmaine, 2022-09-23:
[image]

closed 2022-09-28 by pmaine