pavel-demin/red-pitaya-notes #1065: Question about Red Pitaya Bandwith in Gnuradio

KimlongSeng, 2022-09-22:
### Description of the setup:

 * Device: Red Pitaya 122-16
 * SD card image: red-pitaya-alpine-3.14-armv7-20220322
 * Application: SDR transceiver / SDR transceiver wide
 * Other relevant information: 

### Description of the problem:

I am attempting to configure the device as a spectrum analyzer leveraging the Gnuradio environment with the goal of having an analyzer that functions like the custom Red Pitaya spectrum analyzer app, but would desire the ability for some customization. The Red Pitaya Spectrum analyzer has a frequency range of 0 to 60 MHz with the entire band visible but using the SDR transceiver and gnu radio the I/Q data rate is limited to 1250 kSPS with a sample rate of 125 MS/s so we are unable to view our entire channel spectrum. I am attempting to view cable television signaling which is much wider. I have tried both using osmocom source and red pitaya source, but the bandwidth seems to tie to the sample rate. Is there a way to change the bandwidth to 60 Mhz?

pavel-demin, 2022-09-22:
> Is there a way to change the bandwidth to 60 Mhz?

No, not with the SDR transceiver application.

closed 2022-09-22 by pavel-demin

KimlongSeng, 2022-09-22:
What application do I have to use?

pavel-demin, 2022-09-22:
I do not know.

If you want to continuously stream 60 MHz of bandwidth from the Red Pitaya board to a PC running GNU Radio, you need a board with a faster interface (USB3 or 10-Gigabit Ethernet). So, I think you need to replace the Red Pitaya board with a more suitable device or to adapt your requirements to the capabilities of the Red Pitaya board.