pavel-demin/red-pitaya-notes #1066: ecosystem-0.95-1-6deb253-sdr-receiver.zip did not run on red-pitaya

zkf0100007, 2022-10-18:
### Description of the setup:

 * Device: Stemlab 125-14
 * SD card image: ecosystem-0.95-1-6deb253-sdr-receiver.zip
 * Application: SDR receiver
 * Other relevant information:

### Description of the problem:
I can run ecosystem-0.92-65-35575ed-sdr-receiver.zip which downloaded from https://www.dropbox.com/.
Then I created ecosystem-0.95-1-6deb253-sdr-receiver.zip following the steps described as followed :
```
git clone https://github.com/pavel-demin/red-pitaya-notes
cd red-pitaya-notes
source /opt/Xilinx/Vitis/2020.2/settings64.sh
make NAME=sdr_receiver bit
source helpers/sdr-receiver-ecosystem.sh
```
when I unzipped the generated ecosystem-0.95-1-6deb253-sdr-receiver.zip and copied into TF card, nothing happend after power on.
Did I miss any steps?
Could anyone give me some advice?
Thanks!

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2022-10-18:
Thank you for testing the `sdr-receiver-ecosystem.sh` script.

This script is quite obsolete and I keep it for historical reference.

I have just tried running this script and I see that the `ecosystem-0.95-1-6deb253.zip` file is no longer accessible and the `arm-linux-gnueabihf-gcc` that comes with Vivado/Vitis 2020.1 cannot build statically linked programs.

I think the easiest solution would be to just skip the [SDR receiver](http://pavel-demin.github.io/red-pitaya-notes/sdr-receiver/) application and work with the [SDR transceiver](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver/) application that is more up to date.

zkf0100007, 2022-10-18:
Thanks for your quick reply!
I will follow your advice.

closed 2022-10-18 by pavel-demin