pavel-demin/red-pitaya-notes #1069: Question: adding redpitaya-multichannelanalyser to Alpine pre built image

dc1rdb, 2022-12-25:
Merry Christmas, Pavel!

[redpitaya-multichannelanalyser](https://github.com/pavel-demin/redpitaya-multichannelanalyser) has some nice features like adjustable bincount values and configurable sample times which are quite useful for different scenarios in gamma spectroscopy.

Would it be possible to add this app to your Alpine pre built image?

Alternatively, would it be possible to add these features to [mcph-client](https://github.com/pavel-demin/mcpha-client)?

Many thanks in advance!

pavel-demin, 2023-01-02:
Adding [redpitaya-multichannelanalyser](https://github.com/pavel-demin/redpitaya-multichannelanalyser) to my SD card image is not trivial and I do not plan to do it.

I do not consider the adjustable bincount a useful feature because rebinning is trivially done by any data analysis system used to analyze spectra. Actually I am thinking of removing the histogram part and just keeping the event stream with timestamps like it is done in [pha.tcl](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/mcpha/client/pha.tcl).

Configurable sample times are not an additional feature of [redpitaya-multichannelanalyse](https://github.com/pavel-demin/redpitaya-multichannelanalyse). This functionality is also present in the code available from this repository ([mcpha.tcl](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/mcpha/client/mcpha.tcl)).

dc1rdb, 2023-01-03:
Ok, understood.
Thanks anyway!

closed 2023-01-03 by dc1rdb