pavel-demin/red-pitaya-notes #1070: Vivado 2022.1 open project with TCL Command (xbip_dsp48_macro error)

ByoungjinKim, 2022-12-30:
Hello.

I'm using Vivado 2022.1. I try this project in this version. But I'm not open project.

The address below was used as a reference..
https://github.com/pavel-demin/red-pitaya-notes/issues/1058

sdr_transceiver_122_88 : build ok
sdr_transceiver_emb_122_88 : error

[image]

how can i fix it?

pavel-demin, 2023-01-02:
I think that the easiest workaround at the moment is to use Vivado 2020.2.

Xilinx has removed the DSP48 macro (xbip_dsp48_macro) from Vivado 2021.1 and newer versions.

The new DSP macro has some issues that the DSP48 macro does not.

I have posted a description of these issues on the Xilinx forum:

https://support.xilinx.com/s/question/0D52E00006iHpkX

pavel-demin, 2023-01-04:
I have replaced `xbip_dsp48_macro` with [dsp48](https://github.com/pavel-demin/red-pitaya-notes/blob/master/cores/dsp48_v1_0/dsp48.v) in all projects. I think that it should now be possible to build all projects with Vivado 2022.1.

ByoungjinKim, 2023-01-05:
build test OK.
Thank you.

closed 2023-01-05 by ByoungjinKim