pavel-demin/red-pitaya-notes #1071: How change Alpine Linux IP ?

ByoungjinKim, 2023-01-02:
Communication is not working, so I want to change the IP.

SDRlab board - router - PC : not Ping

SDRlab - PC (direct connect) : Ping OK

i guess gateway problem. so i want change ip.

I connected it to the linux on board via USB.

vi /etc/dhcpcd.conf

change ip address, and then reboot

The ip is not changed, and the modified file is restored.

How can I do it?

[image]

pavel-demin, 2023-01-02:
You are almost there.

You need to run the following command to save the configuration changes to the SD card:
```
lbu commit -d
```

ByoungjinKim, 2023-01-03:
> You are almost there.
> 
> You need to run the following command to save the configuration changes to the SD card:
> 
> ```
> lbu commit -d
> ```

thank you
ip change OK
but, not operating.
I think it's because of security in the company.
direct link is ok

closed 2023-01-03 by ByoungjinKim