pavel-demin/red-pitaya-notes #1072: SDR#, GNU radio not operating, How can i fix it?

ByoungjinKim, 2023-01-03:
HDSDR : OK

SDR# program : no source list
- copyed put ExtIO_RedPitaya_122_88.dll in SDR# folder, but it is not in the source list.

[image]

GNU radio program
- There is an error, can't start
[image]

how can i fix it?

thank you

ByoungjinKim, 2023-01-03:
The GNU radio program identified the problem.
It is not registered in the environment variable.
So in \radioconda\Library\share\gnuradio\grc\blocks folder
After copying .xml and .yml, the error disappeared.

But I don't know about SDR# yet.

pavel-demin, 2023-01-03:
Looking at the SDR# screenshot, I would say that you are not using the SDR# version linked from my notes. As far as I know, newer versions of SDR# do not support ExtIO plugins.

The provided GNU Radio examples only work on Linux.

ByoungjinKim, 2023-01-03:
sorry. Didn't check the version.
Does it only work on Linux?
Um... Then I'll have to make a new Windows version.
Okay. thank you

closed 2023-01-03 by ByoungjinKim