pavel-demin/red-pitaya-notes #1075: Opening the block diagram from tcl file

Jraviteer, 2023-02-02:
### Description of the setup:

 * Device: STEMlab 125-14-Z7020-LN
 * SD card image: red-pitaya-alpine-3.14-armv7-20220322
 * Application: SDR Transceiver
 * Other relevant information: Vivado 2020.2

### Description of the problem:
Dear Pavel,

I've tried to open the block diagram by running the block_design.tcl, through vivado tcl console, and got this message:
[image]

So I tried following some instructions from this thread: https://github.com/pavel-demin/red-pitaya-notes/issues/1058.
But when I run the build-cores.tcl this is the error I get:
[image]

And it is keep telling me to use '-force' command which doesn't work either. And I was wondering if there is a solution to this. I just want to look at the block diagram...
[image]

Many thanks.

Jraviteer, 2023-02-02:
Thank you. Deleting the tmp directory did remove [Common 17-53] error, however, I could not open the block design.

So, I downloaded the source code you have provided, but I'm still getting errors whilst running build-cores.tcl:
[image]

pavel-demin, 2023-02-03:
Since your screenshot does not show the commands that you are running, I cannot say for sure what the problem is. I can only guess that you probably forgot to run the `cd` command.

Here is a copy of the commands from #1058:

**a) Open Vivado, and in the Tcl console, paste this code:**
```
cd c:/#your-path-to-the-code#/red-pitaya-notes-master
source helpers/build-cores.tcl
```
**b) Build the project**
```
set argv [list sdr_receiver xc7z010clg400-1]
source scripts/project.tcl
```
If these commands do not work for you, I can only recommend using WSL (Windows Subsystem for Linux), installing [Ubuntu 20.04](https://apps.microsoft.com/store/detail/ubuntu-2004/9N6SVWS3RX71) from Microsoft Store and installing Xilinx tools for Linux. This way you can use all the scripts in this repository and follow the examples in the notes.

closed 2023-02-05 by pavel-demin

Jraviteer, 2023-02-08:
Sorry, for the late question, but does it have to be Ubuntu 20.04? because I am currently using Ubuntu 22.04.1.

pavel-demin, 2023-02-09:
I think that Ubuntu 22.04 should be OK.

Yes, the part number should match the one installed on your board.

Jraviteer, 2023-02-10:
It worked. Thank you so much!