pavel-demin/red-pitaya-notes #1078: Buffer modules

pasicl, 2023-02-03:
### Description of the setup:

 * Device: 125-14-Z7020-LN
 * SD card image: red-pitaya-alpine-3.14-armv7-20220322
 * Application: pulsed_nmr
 * Other relevant information: Vivado 2017.2

### Description of the problem:
Dear Pavel, 
I am trying to build the project in Vivado, and when I synthesize the project, I recieve an error message, which says that the modules input_buffer and output_buffer are not in the project. I built the cores as specified in the path helpers/build-cores.tcl, but in this TCL script it doesn't build these buffers, hence the project cannot find them, and there is no "core-config.tcl" file as with the other cores. 
How can I include them?

[image]
Thank you very much for all your work!

pavel-demin, 2023-02-03:
Please provide a list of commands that I could use to reproduce this problem.

The modules from `cores/common_module` are added to the project by the `project.tcl` script. If you are not using Makefile or the `project.tcl` script, then you will have to somehow add the missing files to your project.

pasicl, 2023-02-03:
What I did was: first of all create the cores with the **create_cores.tcl** script, but this scrips does not include the modules in the folder **cores/common_modules**, which I guess is the problem.
Then I use the **project.tcl** to create the project. Once I am in Vivado, when I try to synthesize the project, the error which I told you earlier appears.

pavel-demin, 2023-02-03:
> ... which I guess is the problem.

No, this is not a problem. This is how it should be. Unfortunately, Vivado does not allow the same Verilog module to be included in several different IP cores.

I have just cloned the latest version of the code from the `master` branch of this repository and created a project with the following commands:
```
source helpers/build-cores.tcl
set argv [list sdr_transceiver xc7z010clg400-1]
source scripts/project.tcl
```
When I open the project, the `input_buffer` and `output_buffer` modules are displayed in the sources tab as can be seen in the following screenshot:

[image]

When I click on "Generate Bitstream", everything works without problem.

I have updated the "master" branch of the repository earlier today and moved all breaking changes to the "develop" branch. Maybe re-downloading the code from the "master" branch might solve your problem.

pasicl, 2023-02-04:
Thank you, I do not know what was I doing wrong (maybe I had deleted some files by mistake) but now it synthesizes without problem.
I have another doubt regarding this project, the pulsed_nmr. I don't have much experience with the Red Pitaya and I'm learning, but it may seem trivial to you. I've taken a look at the server.c code and cannot understand how it writes to the registers in the FPGA, which for this project are cfg_0, rx_0/reader_0, tx_0/reader_0 and sts_0. 
[image]

I see that at the beginning of the code you map these registers to the 4 associated pointer variables. Does that mean that every time this variable is modified, the data is sent by the processor through the AXIS protocol to the registers?

Thank you very much for your help!

closed 2023-02-05 by pavel-demin