pavel-demin/red-pitaya-notes #1083: Building a bootable SD card image: fails by missing programms

waterwin, 2023-03-01:
### Description of the setup:

 * Device:
 * SD card image:
 * Application:
 * Other relevant information:

### Description of the problem:

### Steps to reproduce the problem:

 1.
 2.
 3.

closed 2023-03-01 by waterwin

pavel-demin, 2023-03-01:
> you can not run a command to install sudo with the command sudo

Thanks. I have updated the notes and removed `sudo` from the `apt-get` commands.

waterwin, 2023-03-01:
libtinfo5needs to be installed before Vitis will install, this is a known Vitis issue, if not it will hang indefinitely on the action " Generating installed device list"
So this pre-req is in fact for the page before : Setup Development Machine / 
Installing Vitis Core Development Kit

Thanks Pavel, as always.
ErwinPE3ES / F4VTQ

pavel-demin, 2023-03-02:
All these packages are installed by the [debian.seed](https://github.com/pavel-demin/red-pitaya-notes/blob/gh-pages/etc/debian.seed) script used to automatically create a virtual machine in the Development Machine notes.