pavel-demin/red-pitaya-notes #1084: Connecting Red Pitaya to Vivado for debugging

nsparks-per, 2023-03-02:
### Description of the setup:

 * Device: STEM Lab 125-14
 * SD card image: Alpine 20220302
 * Application: sdr-transceiver
 * Development Machine: 
     * VirtualBox running on windows 10
     * Debian 11 with xfce4 GUI
     * Vivado 2020.2

### Description of the problem:

Hey,

I opened a thread a couple months ago asking if the transmit and receive channels where capable of being phase synced and you gave me some advice on how to modify the vector analyzer fpga image. I've been working on doing this but I've run into a development process issue. So far I've been modifying the image in Vivado on a virtual machine. Using the make command to create a .bit file. Then copying the .bit file to the red pitaya over the network using SCP. This has been working so far and I was able to modify the output on the DACs to be the same sinusoid. I'm now trying to modify the receive side and it's proving to be more difficult. 

I would like to use the Vivado debugger to debug the Red Pitaya but with Vivado being on a virtual box machine I can't use the USB ports without installing the Virtual Box extended capabilities package. The issue with this is I've found it causes my virtual OS to no longer boot and made me start from scratch with a new virtual machine. Is there a specific method you used to debug your projects that gets around this?

One way to get around the virtual machine issue is of course creating an actual machine. I'm not sure if I would be able to run your boot setup script as I don't believe the actual computer would have internet access until after the OS installs drivers and such. Is there anything you see that could get in my way if I tried to set up either a machine running Debian 11 or possibly Ubuntu?

I've really appreciated how much work you've put into this. Hopefully, you can point me in a good direction to get my own development process working smoothly.

Thanks,

Nick

pavel-demin, 2023-03-02:
> Is there a specific method you used to debug your projects that gets around this?

I just use printouts to figure out what is going on.

> Is there anything you see that could get in my way if I tried to set up either a machine running Debian 11 or possibly Ubuntu?

Both Debian and Ubuntu should be OK.

Another option could be using WSL (Windows Subsystem for Linux) and installing [Ubuntu 20.04](https://apps.microsoft.com/store/detail/ubuntu-2004/9N6SVWS3RX71) from Microsoft Store.

nsparks-per, 2023-03-02:
> I just use printouts to figure out what is going on.

Could explain this a bit more? It sounds like your referring to printing to the console when debugging the C side of things. I'm trying to find the error in my Verilog program. 

>Another option could be using WSL

so I would have Vivado installed in windows and then use WSL for building the projects and such?

pavel-demin, 2023-03-02:
It is indeed for the C debugging.

For Verilog, it is recommended to write tests, run them in a simulator and look at the timing diagrams. It is also possible to add printouts to Verilog code and tests. The [cocotb](https://www.cocotb.org/) library is quite popular now and I also use it to write some of my tests.

I proposed to use WSL as a replacement for the VM and install Vivado and everything under WSL/Ubuntu.

Installing Vivado in Windows is yet another option.

closed 2023-03-11 by pavel-demin