pavel-demin/red-pitaya-notes #1085: SDRLab 122-16

Dystroxic, 2023-03-07:
### Description of the setup:

 * Device: SDRLab 122-16
 * SD card image: RedPitaya OS 1.04-15 (version that shipped with the board)
 * Application: sdr-transceiver

### Description of the problem:

I'm getting a "Connection Refused" error when trying to connect to my board. A couple questions that could clarify what the issue is:

- Do your blocks and this application work with the image that is shipped with the SDRLab 122-16? Or, is your image customized in a way that is necessary for your applications to function?
- Does your custom image work with the SDRLab 122-16? On [your README page](https://pavel-demin.github.io/red-pitaya-notes/), this model isn't listed, it only lists the STEMLab models.
- I'm able to access the dashboard hosted by the SDRLab (so I know the IP is correct and accessible). I see there's a "Data Stream Control" page that appears to set up the board for streaming over TCP/IP, and that seems to be what I'd want to do with GNURadio. Do I need to start the streaming in order for this to work?

### Steps to reproduce the problem:

 1. Use an SDRLab 122-16 with the factory image
 2. Follow this repository's instructions for running the `sdr-transceiver` application
 3. Update the flowgraph with the correct IP
 4. Get a `ConnectionRefusedError: [Errno 111] Connection refused` error

pavel-demin, 2023-03-08:
> 1. Use an SDRLab 122-16 with the factory image
> 2. Follow this repository's instructions for running the sdr-transceiver application

As far as I know, there are no instructions in this repository on how to run the `sdr-transceiver` application on the SD card image provided by Red Pitaya. So I do not understand what instructions you are following.

> Does your custom image work with the SDRLab 122-16? On [your README page](https://pavel-demin.github.io/red-pitaya-notes/), this model isn't listed, it only lists the STEMLab models.

When I wrote the notes, the SDRLab 122-16 board was called STEMlab 122.88-16 SDR. So yes, SDRLab 122-16 is supported by my custom image. I have just updated the notes index page and added SDRLab 122-16.

The `sdr-transceiver` application is for the 125-14 board, not the 122-16 board. When you run the `sdr-transceiver` application on the 122-16 board, the system freezes completely and that is why the connection does not work.

The correct application for the 122-16 board is `sdr-transceiver-122-88`. Notes on this application and installation instructions can be found at the following link:

http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-122-88/

> Do your blocks and this application work with the image that is shipped with the SDRLab 122-16? Or, is your image customized in a way that is necessary for your applications to function?

Some of my applications do indeed use some functionality that is not available on the SD card image provided by Red Pitaya. I think the `sdr-transceiver-122-88` application should work on the SD card image provided by Red Pitaya after recompiling the C code.

Dystroxic, 2023-03-08:
Thank you for the quick response. I have now done the following:
- Loaded an SD card with your `red-pitaya-alpine-3.14-armv7-20220322` image
- Copied the `apps/sdr_transceiver_122_88/start.sh` file to the root of the SD card
- Started the board with the card. Able to ping it, so it's running.
- Run `gnuradio-companion trx_am.grc`

Now I'm getting the following errors:
```
Executing: /usr/bin/python3 -u /home/me/red-pitaya-notes/projects/sdr_transceiver_122_88/gnuradio/trx_am.py

ALSA lib confmisc.c:855:(parse_card) cannot find card '0'
ALSA lib conf.c:5178:(_snd_config_evaluate) function snd_func_card_inum returned error: No such file or directory
ALSA lib confmisc.c:422:(snd_func_concat) error evaluating strings
ALSA lib conf.c:5178:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1334:(snd_func_refer) error evaluating name
ALSA lib conf.c:5178:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5701:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2664:(snd_pcm_open_noupdate) Unknown PCM default
audio_alsa_source :error: [default]: No such file or directory
Traceback (most recent call last):
  File "/home/me/red-pitaya-notes/projects/sdr_transceiver_122_88/gnuradio/trx_am.py", line 344, in <module>
    main()
  File "/home/me/red-pitaya-notes/projects/sdr_transceiver_122_88/gnuradio/trx_am.py", line 322, in main
    tb = top_block_cls()
  File "/home/me/red-pitaya-notes/projects/sdr_transceiver_122_88/gnuradio/trx_am.py", line 225, in __init__
    self.audio_source_0 = audio.source(samp_rate, '', True)
RuntimeError: audio_alsa_source

>>> Done (return code 1)
```

From some quick searching, it seems like this is an issue with not having a sound card. I'm running `gnuradio-companion` in a virtual machine with no sound card; is this not supported? Is there any other way to run it from a VM?

Specifically, I'm running it in WSL2.

pavel-demin, 2023-03-08:
> I'm running gnuradio-companion in a virtual machine with no sound card; is this not supported?

I think that it is supported. You will just need to remove the audio sink from the flow graph or use [trx_dual_template.grc](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_122_88/gnuradio/trx_dual_template.grc) that does not have any audio block.

Dystroxic, 2023-03-10:
Thank you, I managed to get it working. For some reason I didn't realize it had the audio sink/source; I was looking at them like a virtual sink/source.

I was hoping you could clarify some parameters though, when it comes to the SDRLab 122-16 specifically. I'm trying to get it working with the [GNURadio FSK packet transceiver demo](https://github.com/duggabe/gr-control/blob/main/Transmitters/pkt_fsk_xmt.grc). Specifically, I'm trying to determine how to adjust the parameters (`vco_max` and VCO `amplitude`) to be suitable for the 122-16 and to maximize the transmit power.

In my setup, I have two [YouLoop antennae](https://airspy.com/youloop/) attached to the 122-16, and with some customization, the packet FSK demo I linked above works as expected. However, it only works when the two antenna are less than 2 inches apart; any more than that, and it can't capture the packets, so I imagine I may not be utilizing the 122-16's full transmit power (which, I admit, is very low, but should still do better than a 2-inch airgap).

pavel-demin, 2023-03-11:
I would start by checking the output signal with an oscilloscope or a spectrum analyzer. Maybe the signal is too weak.

What frequencies do you use? Are they above 300 kHz?

Dystroxic, 2023-03-11:
I'm currently testing with 6.780MHz.

Is it the case that with your SDRLab libraries/blocks, an output amplitude of `+1.0` translates to 100% transmit power? That seems to be the convention with most SDR sinks, but since yours is custom-made, I can't find documentation on this.

pavel-demin, 2023-03-11:
The frequency looks OK.

The maximum amplitude is 2.0. There is a factor of two lost somewhere in the DSP blocks of the FPGA configuration. Still to be improved.

Dystroxic, 2023-03-11:
That would explain it. Most examples use an amplitude of 0.75 or so to prevent saturation, so I'm transmitting at about 38% power!

Dystroxic, 2023-03-11:
Is there any way to increase the receive gain on the SDRLab?

pavel-demin, 2023-03-11:
I think the general practice is to increase the transmit gain with a power amplifier and an antenna and increase the receive gain with an antenna and a preamplifier. I do not see why it would not work with the SDRLab board.

I would still check the output signal with an oscilloscope or spectrum analyzer or other suitable analyzers. Maybe you will notice some issues that could be fixed in the software.

Dystroxic, 2023-03-11:
I was asking because most of the SDRs, even those much less expensive than the SDRLab, have programmable gain. For example, the [HackRF One](https://hackrf.readthedocs.io/en/latest/faq.html#what-gain-controls-are-provided-by-hackrf) has RF, IF, and Baseband Rx gain values that can be set in the GNURadio block.

I was just hoping that the SDRLab had something similar.

pavel-demin, 2023-03-11:
Since I have answered all the questions and you have a setup that seems to work, I am closing this issue.

closed 2023-03-11 by pavel-demin