pavel-demin/red-pitaya-notes #1086: Scanner image tcp port closed

mihmoz, 2023-03-08:
### Description of the setup:

 * Device: RP 125-14
 * SD card image: red-pitaya-alpine-3.14-armv7-20210701-scanner.zip -- red-pitaya-alpine-3.14-armv7-20220322-scanner.zip
 * Application: scanner
 * Other relevant information:

### Description of the problem:
Hi Pavel, I seem to run into an issue with the scanner images:
The scanner.py application says "connection refused".
nmap says port 1001 is not open.
All scanner image versions after 20210701 are affected on my board. Previous versions seem to work.
Dmesg on the RP with the affected image ends with these lines:
macb e000b000.ethernet eth0: configuring for phy/rgmii-id link mode
macb e000b000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
urandom_read: 3 callbacks suppressed
random: dhcpcd: uninitialized urandom read (112 bytes read)
random: crng init done
Alignment trap: not handling instruction f8cb3001 at [<0047076e>]
8<--- cut here ---
Unhandled fault: alignment exception (0x811) at 0xb6f47001
pgd = 1f6af58f
[b6f47001] *pgd=06a8a831, *pte=40000743, *ppte=40000c33

### Steps to reproduce the problem:

 1. 
 2.
 3.

pavel-demin, 2023-03-11:
Thank you for testing the scanner images.

I have just checked the latest release and the scanner program does indeed crash on startup. I also noticed a similar issue with some other applications when upgrading to Alpine Linux 3.14 and GCC 10.3.

Looks like GCC 10.3 is doing something wrong when it optimizes access to the 8-bit configuration registers.

The easiest workaround at the moment is to recompile `scanner.c` without the `-O3` flag.

I will check if the problem persists with newer versions of Alpine Linux and GCC, and then I will try rearranging the configuration registers.

pavel-demin, 2023-08-10:
The fixed version of the scanner application is now in the [new release](https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20230705).

closed 2023-08-10 by pavel-demin