pavel-demin/red-pitaya-notes #1087: Data types and transfer in pulsed_nmr

pasicl, 2023-03-10:
Description of the setup:

Device: 125-14-Z7020-LN
SD card image: red-pitaya-alpine-3.14-armv7-20220322
Application: pulsed_nmr
Description of the problem:

I am trying to use both ADCs of the StemLab with the pulsed_nmr application, and I don't understand how is the data transfer from the FPGA to the server software running in the Red Pitaya. I've tried using both channels and only the first one seems to work. 

From what I understand looking at the block diagram of the FPGA project, there are 4 streams of 32 bits each one: 1st ADC (I), 1st ADC (Q), 2nd ADC (I), 2nd ADC (Q). Then they are combined with the stream combiner and it is a unique stream of 128 bits. Then it enters a FIFO which allows to write in 128 bits at a time but only read 32 bits, and these 32 bits are stored in the Stream-reader, which is associated to the register with starting adress 0x40010000. Does this mean that only 1 of of the 4 streams is used?
[image]

Then, from the C code, it stores in a buffer the contents of such register, and sends it to the client, which uses a uint64_t variable to store the data. However, in the register, the data is only 32 bits long. I tried reducing the size of the *rx_data variable and the *buffer and it seems to work the same way, and even this 32 bit long data contains real and imaginary parts. 

What am I missing? I don't get the format of the streams and how they are transferred. Is there any way to debug the FPGA (I don't know any method) or to access the data inside the buffer in the c-code?

pavel-demin, 2023-03-11:
The sample type is a 32-bit signed integer. When copied to memory, the I and Q samples are interleaved exactly as you describe: 1st ADC (I), 1st ADC (Q), 2nd ADC (I), 2nd ADC (Q), etc.

pavel-demin, 2023-03-11:
BTW. Since you are looking at the code, I recommend looking at the code in the [develop branch.](https://github.com/pavel-demin/red-pitaya-notes/tree/develop/projects/pulsed_nmr). There are some improvements and some of them may fix the #1080 issue.

pasicl, 2023-03-16:
Perfect, that is exacly what I needed, thank you!

closed 2023-03-16 by pasicl