pavel-demin/red-pitaya-notes #1091: Issue with starting sdr_transceiver_hpsdr_122_88 on the TRX-DUO with 122.88 clock

enthru, 2023-04-08:
* Device: TRX-DUO
 * SD card image: Your latest image
 * Application: sdr_transceiver_hpsdr_122_88

Hello Pavel! 
Now i've replaced generator with crystek cvhd-122.88 and TRX-DUO just being unresponsive when i'm starting 122_88 version of the app.
125MHz version starts okay, so generator is working. But of course it's unusable as clock is different.
Is it possible do debug it somehow?

pavel-demin, 2023-04-08:
As I already mentioned in your previous issue, the main difference between the 125-14 and 122-16 boards is the type of FPGA. So, a bistream file for one board cannot be used for the other board.

There is nothing to debug. You will need to create a new project that works with a 122.88 MHz generator and is compatible with the `xc7z010clg400-1` chip.

closed 2023-04-08 by pavel-demin

enthru, 2023-04-08:
Sorry, i'm pretty new with FPGA, so missed that fact that bitstream cannot be used for the other chip (but yeah it should be pretty obvious). Thanks again.

pavel-demin, 2023-04-08:
I think that it should be possible to build a bitstream file that might work with your modified board using the following command:
```
make NAME=sdr_transceiver_hpsdr_122_88 bit
```

enthru, 2023-04-08:
It works :) Thanks!