pavel-demin/red-pitaya-notes #1094: GNURadio Red Pitaya block's "Freq. Correction PPM" field has no effect

Dystroxic, 2023-05-15:
### Description of the setup:

 * Device: SDRLab 122-16
 * SD card image: red-pitaya-alpine-3.14-armv7-20220322
 * Application: SDR Transceiver
 * Other relevant information: Using GNURadio

### Description of the problem:

The "Freq. Correction ppm" field on the Red Pitaya Sink and Source blocks appear to have no effect. With a simple flowgraph of Signal Source -> Red Pitaya Sink, and Red Pitaya Source -> GUI Sink, we can see that changing the `ppm` value on either block (either in the flowgraph in advance, or at runtime) has no effect.

### Steps to reproduce the problem:

 1. Run [this flowgraph](https://github.com/pavel-demin/red-pitaya-notes/files/11481818/ppm_test.grc.txt) (change file extension first)
 2. The center frequency is set at 500.0 MHz and the `ppm` on the source is set at 200, so the GUI frequency sink should show the signal at 500.1MHz, but it shows it at exactly 500.0 MHz.
 3. Try setting the `ppm` on the Sink block, and you'll see that it also has no effect.

pavel-demin, 2023-05-16:
Thank you for finding this problem. 

I have never used the frequency correction part of the blocks and recently thought about removing it.

I think it would be easier to add the following expression directly to the frequency field:
```
freq * (1.0 + 1e-6 * corr)
```

Dystroxic, 2023-05-16:
I tried that as well, but there are very odd behaviours when that field is changed live during runtime. I ended up using a `Frequency Shift` block before the sink and after the source, which works as expected with runtime changes.

pavel-demin, 2023-07-03:
I have just tested your flow graph and I think that the problem is not that the frequency correction does not work but that the 500 MHz frequency is too high.

I have added the "QT GUI Range" block to the flow graph to control the frequency correction and I can change the frequency correction in real time without any problem if I set the center frequency to 10 MHz.

The SDR transceiver application uses 28 bits for all values sent from client to server. The maximum frequency is therefore about 268 MHz. The sampling rate of the Red Pitaya board being 125 MSPS, it is more than enough to cover the frequencies of the first Nyquist zone.

I would say that the solution to your problem is to convert the 500 MHz frequency to the frequency of the first Nyquist zone before sending it to the Red Pitaya blocks.

closed 2023-07-03 by pavel-demin