pavel-demin/red-pitaya-notes #1095: Question on pulse length dependency for the pulse height measurement

mkoehli, 2023-05-17:
We are using the Red Pitaya (125-14) since more than a decade.
We are using it for reading out pulse heights from proportional counters. Typical pulses are in the order of 500 ns to 20 mus and look like a Gamma pulse. There is of course a certain amount of noise on the baseline.
Recently, we upgraded the old version of the OS from more than 10 years ago to the most recent stable, including upgrading the MCPHA.

We noticed the following behavior after changing:
- The baseline auto correction seems to not work properly any more. We get a significant offset of sveral thousand channels. In the old version from 10 years ago such behavior appeared for example if the baseline correction was not used. 
- The pulse height measurement seems to depend on the pulse length. Practically, for pulses shorter than 20 mus the measured height depends on the pulse width, above that it seems to measure the same height, regardless of the pulse length (as it should be). Have there been any significant changes on the FIR filters which now produces a behavior similar to a low pass filter? If so, is there any way to change that back to as it was before, even if it would mean that the performance for long pulses would be worse?

Any help is appreciated.

pavel-demin, 2023-05-17:
I am glad to know that you have been using the MCPHA application for so many years.

Do you have the version number of the MCPHA application you were using before the update? This information would help me understand the problem.

From your description of the problem, I would say that the version of your control program is not compatible with the version of the MCPHA application running on the Red Pitaya board.

I changed the codes for some commands in January and December 2016 with following commits:
https://github.com/pavel-demin/red-pitaya-notes/commit/5877976bdbe870a020730d8a725681b853d2f19e
https://github.com/pavel-demin/red-pitaya-notes/commit/f0fd92d4c318017582164714f891e27ea5683086

Could you please check that the version of the control program is the same as the version of the MCPHA application running on the Red Pitaya board?

mkoehli, 2023-05-17:
In order to find out the exact version back then I'd probably need to reach out to the person who originally compiled that version back then. I can only tell for sure that it was not modified any more since June 2016 when the original Red Pitaya was replaced by a new one. But the OS and MCPHA versions we had back then in 2016 were compiled from a virtual machine which was significantly older.

The current version, however, is the latest stable which was obtained from GitHub (Red Pitaya OS, MCPHA application and MCPHA Client). And to make clear, we are not interchanging old and new versions. We just didn't change anything since then.

The commit from January 2016 "add controls for baseline and max threshold to mcpha server and client" does not load, but I can tell that the MCPHA client from 2016 we used already had the baseline substraction (auto/manual).

pavel-demin, 2023-05-17:
> The current version, however, is the latest stable which was obtained from GitHub (Red Pitaya OS, MCPHA application and MCPHA Client).

Unfortunately, that is not enough information for me. Having the exact version numbers would really help.

Would it be acceptable for you to install both the MCPHA application and the MCPHA client by following the getting started instructions from the following link?

http://pavel-demin.github.io/red-pitaya-notes/mcpha/

This way we will be sure to talk about the same version.

mkoehli, 2023-05-17:
Sure, I will reinstall MCPHA.

Just as a sidenote: It would be very useful to have version number and a release date presented somewhere in the web interface and in the application (and not somewhere hidden in the source code). For example an "about" button could do the job.

mkoehli, 2023-05-22:
Actually, reinstalling solved the problem.

Currently, I am not sure what was the reason for that behavior described above. I cannot tell whether the former version was the absolutely latest or a prior build. With simply both the latest it worked. I also encountered some errors on the SD when trying to rewrite it, and there could be an issue linked to that, but it must have been quite unlucky to exactly cause such.   
The second issue I figured is due to RedPitaya's FPGA-based anaylsis, so that's an 'internal' problem.
So, it seems everything is solved as far as it could.

closed 2023-05-22 by mkoehli